randdusing/cordova-plugin-bluetoothle #505: iOS App stucks by reading characteristics

RitterD, 2018-10-10:
Hi,

first of all my app works perfectly on android but stucks at a certain point on iOS.
There are some characteristics which are unable to read as you can see in the screenshot:
[image]

This screenshot is from the android app. After the read sequence it continues smoothly.

But in the iOS app it says "readSucces" at the same characteristic but stucks at this point...
[image]
 
I used your example:
        readSequence = readSequence.then(function () {

            return new Promise(function (resolve, reject) {

                bluetoothle.read(resolve, reject,
                    { address: address, service: serviceUuid, characteristic: characteristic.uuid });

            }).then(readSuccess, handleError);

        });
    });
    return readSequence;

Is it possible to interrupt the read sequence or skip the characteristics where the code stops working?

Thank you very much in advance

RitterD, 2018-10-11:
Update:
I found a way to bypass the issue I suppose.
I call the addvertising "bluetoothle.startAdvertising" after the characteristic "2A52" is read.

But the advertising fires me this error:

2018-10-11 11:52:31.423001+0200 smart[1696:690224] *** WebKit discarded an uncaught exception in the webView:decidePolicyForNavigationAction:request:frame:decisionListener: delegate: <NSInvalidArgumentException> *** -[__NSPlaceholderDictionary initWithObjects:forKeys:count:]: attempt to insert nil object from objects[0]
2018-10-11 11:52:31.423205+0200 smart[1696:690271] Client application failed to make a policy decision via WebPolicyDecisionListener, using defaultPolicy 2

Did you ever get this error when you call the advertising function?

RitterD, 2018-10-11:
Update:
If s.o runs in this problem just skip the "bluetoothle.startAdvertising" and go direktly to the subscribe mehtods.
I honestly dont know if the andorid app still wokrs know, but for iOS this solved the issue.

closed 2018-10-11 by RitterD

randdusing, 2018-10-12:
Reading and advertising should have nothing to do with each other...

RitterD, 2018-10-12:
It's weird... the "bluetoothle.isAdvertising" returns me "false" but subscribing works. I remember that on Android development I couldn't subscribe without starting the advertising method. I'm going to test the app on android in the following days.