randdusing/cordova-plugin-bluetoothle #508: Windows : unable to close

FredGr, 2018-11-01:
On windows 10 when I close the BLE connection, the device remains active.
I looked inside the plugin code and it looks like services are being closed but not the actual device.
Also the success callback is never called.

TheBosZ, 2018-12-03:
I've started a rework of the Windows 10 support here: https://github.com/TheBosZ/cordova-plugin-bluetoothle/tree/thebosz/rewrite-windows-10

Close() now works, and it uses MAC addresses as identifiers.

I'll be working on it more as Windows 10 support for our app has been identified as a priority.

FredGr, 2018-12-04:
Hello,
Thanks for this very good initiative.
I managed to solve the closing issue by adding 'device.close() to the close function in BluetoothLEPlugin.js such as 👍       

var device = cachedDevice;
      device.close();
      device = null;
      var result = {status : "disconnected"};
      successCallback(result);
    
    } else {
      errorCallback({ error: "close", message: "Invalid parameter" });
    }
and adding cachedDevice initialization line 277
    cachedDevice = bleDevice;

Having done that the plugin is working quite well on Windows10 (connect read and write and close)

closed 2018-12-25 by randdusing

randdusing, 2018-12-25:
Please reopen if @TheBosZ 's recent changes didn't fix this.

PR #511 (2018-12-25): Rewrite Windows 10 support
This PR is to bring Windows 10 UWP support up to about the same level as Android and iPhone.

I did what I could to make the experience the same between the platforms.

We're using this plugin in an app that's targeting Android, iOS and Windows. With this PR, the app works as expected across the three.

The major missing functionality is advertisement data. I'm still working on that, but right now it appears as though the end-user has to jump through a lot of hoops to get it working so it may not be viable.

This closes #487, closes #491, closes #501, closes #508, an [... cut]