randdusing/cordova-plugin-bluetoothle #513: How do I get the service?

RZR666, 2018-12-20:
I believe it's encrypted in the result.advertisement but I'm having an issue trying to decode that?

I've used bytesToEncodedString and bytesToString but neither are giving me anything to work with?

Any help appreciated. 

Thanks

randdusing, 2018-12-21:
You want to use #encodedStringToBytes() as described here: https://github.com/randdusing/cordova-plugin-bluetoothle#startscan

RZR666, 2018-12-21:
Yeah sorry that's the one I'm trying but it's giving me,

2,1,6,3,3,170,254,21,22,170,254,32,1,18,71,218,130,102,46,79,165,228,189,106,84,18,108,52,250,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0

randdusing, 2018-12-25:
So that's the list of bytes. What to actually do with it depends on the format of the advertisement data.

AdrGeorgescu, 2019-05-20:
how do you read the advertisement on android devices?
how do you decode that parameter?
``` advertisement = advertisement data in encoded string of bytes, use bluetoothle.encodedStringToBytes() (Android) ```

if I use bluetoothle.encodedStringToBytes() I get an array with bytes. How can I decode this information?

For example on iOS I can read the "localName" parameter. how can I do that on Android?

ndrake, 2019-05-20:
This page gives and overview of BLE advertising: 

https://www.silabs.com/community/wireless/bluetooth/knowledge-base.entry.html/2017/02/10/bluetooth_advertisin-hGsf

There are flags in the data that mark each component, for example: 0x08 is the flag for "shortened local name".

AdrGeorgescu, 2019-05-24:
For example I get a list of bytes [0, 0, 0, 70]. This is the value for signal strength, an uint16 value.
how can this be transformed to a human readable value? An integer value with radix 10?

ndrake, 2019-05-24:
We use the following ```arrayToHexString``` function to convert the Uint8Array returned from ```encodedStringToBytes()``` to a string of hex.  

```
  private arrayToHexString(hexData : Uint8Array) {
    let hexString : string = '';
    for(let i = 0; i < hexData.length; i++) {
      hexString += this.hexToString(hexData[i]);
    }
    return hexString;
  }
  
  private hexToString(hex : number) : string {
    let hexString : string = hex.toString(16);
    if (hexString.length % 2) {
      hexString = '0' + hexString;
    }
    return hexString;
  }
```

Then we grab numbers from the hex string with something like this:

```Number.parseInt(hexString.substring(4, 6), 16)```