randdusing/cordova-plugin-bluetoothle #522: Enable function callback on Android

UlisesCeca, 2019-01-22:
Hello, I am trying to call the `enable` function but somehow it's never getting into its callback functions. The only way I got to turn on the bluetooth is the following:

```
this.platform.ready().then(() => {
      bluetoothle.initialize(res => {
        console.log(res)
        bluetoothle.enable(aa => {
          console.log("entra")
        }, error => console.log(error));
      })
    })
```

This works and turns on the blueetoth on some devices, but it never gets into the callbacks of `enable`. Why? Which one is the correct to use?

Also, there are some smartphones that ask you if you want the App to turn on the blueetoth and opens a dialog to turn it on or not. Is there a way to detect what the user chooses when this dialog shows up in some devices?

Thanks

aroncal, 2019-01-22:
You could use just the "initialize" method with the "request" paramater set to true. That way, I think that if the system prompts the user to activate the Bluetooth with a popup, the callback waits until the result of that action, whether it's "enabled" or "disabled".

UlisesCeca, 2019-01-22:
Thanks, this seems to be the way. However, if I don't allow the App to turn on the bluetooth in the dialog, it won't let me open that dialog again unless I restart the App. Do you know something about this?

aroncal, 2019-01-22:
If the bluetooth is disabled, you can present an informational message to the user BEFORE calling "initialize", explaining why the App needs Bluetooth. After that it would be the user's concern to activate it manually if he/she didn't activate when asked. That's is the way I did.

Otherwise, I suppose you could call "enable" (android only) or "initialize" again. You have to figure out the best workflow for your use case.

mgplanas, 2020-06-29:
What I do is to modify Plugin (BluetoothLEPligin.java) Because I think it misses this 3 lines before the enable statement (line 1058):

    //Get Bluetooth adapter via Bluetooth Manager
    Activity activity = cordova.getActivity();
    BluetoothManager bluetoothManager = (BluetoothManager) activity.getSystemService(Context.BLUETOOTH_SERVICE);
    bluetoothAdapter = bluetoothManager.getAdapter();
    boolean result = bluetoothAdapter.enable();

After inserting this lines I remove and add the android platform and everything works. I hope these modifications will be included in the next release.

mgplanas, 2020-06-29:
I've just created a pull request with these modifications. I hope they include them.