randdusing/cordova-plugin-bluetoothle #523: Android: "Unable to write on return" with status 129 when reconnecting

MalikDrako, 2019-01-29:
I am working on OTA updating a BLE device, and am having problems with some phones. The device has a service with two characteristics: a write-only characteristic for commands, and a subscribable characteristic for replies. 

On some phones, the update works fine, but on others it fails to reconnect, giving "Unable to write on return". When I attach Android Studio, I see status 129 (`GATT_INTERNAL_ERROR`) in `onCharacteristicWrite`

Update process:

- Write Start Update command
- Device sends ACK reply then resets into bootloader (If no data is received, the bootloader exits after 5s)
- Phone disconnects, waits ~1s to be sure device has reset (I tried increasing to ~3s with no change)
- Phone connects, discovers, subscribes (I tried scanning while this is happening on Android)
- Phone starts handshake ("Unable to write on return" with status 129 on the first write here, then ~15s later I get "Device is disconnected")
- Phone sends firmware packets
- If successful, phone sends completion command

I looked at the HCI log, but it looked like the phone didn't actually connect to the device, just `Set Scan Enable` and related, then `Disconnect Complete` ~19s later

Strangely, "Recovery Mode" updates work fine (the device resets into the bootloader for 5s when it is plugged in): 

- Disconnect if connected
- Scan for a device in recovery mode
- Go to "Phone connects, discovers, subscribes" in the normal update process

untilbit, 2019-06-03:
Did you fix this issue?

Funkelfetisch, 2021-08-19:
same, any update on this?

clarklight, 2021-09-14:
I am seeing the same issue as well, but i am using 2 android phone, 1 acting as a peripheral device. The device that write to the phone is showing Unable to write on return in the error message, and when i do a read(), its also showing unable to read.