randdusing/cordova-plugin-bluetoothle #527: Android to Android Connection???

AkashRajpurohit, 2019-02-26:
I wanted to know if this plugin helps to create a connection between two android devices via Bluetooth.

I have tried implementing it but it doesn't return any devices for the startScan() Observable
I have requested the location permission beforehand and provided the permissions as well, still now luck.

Here is my code in the constructor
```typescript
constructor(
    public navCtrl: NavController,
    private bluetooth: BluetoothLE,
    private platform: Platform
  ) {
    this.platform.ready().then(async () => {
      const { status } = await this.bluetooth.initialize()
      if (status !== 'enabled') {
        alert('Bluetooth in not enabled. Please enable it')
      }
      const { hasPermission } = await this.bluetooth.hasPermission()
      if (!hasPermission) {
        await this.bluetooth.requestPermission()
      }
      const { isLocationEnabled } = await this.bluetooth.isLocationEnabled()
      if (!isLocationEnabled) {
        await this.bluetooth.requestLocation()
      }
    })
  }
```

The startScan function looks like this
```typescript
startBluetoothScan() {
    const scanOpts: ScanParams = {
      allowDuplicates: true,
    }
    try {
      this.bluetooth.startScan(scanOpts).subscribe(
        res => {
          alert(JSON.stringify(res))
        },
        err => {
          alert(JSON.stringify(err))
        }
      )
    } catch (e) {
      alert('Error in start scan')
      alert(JSON.stringify(e))
    }
  }
```

When i call this function on button click i get the first alert res of 
```
{ "status": "scanStarted" }
```
But no devices are shown later even though it is subscribed to the observable.
Both startScan and stopScan works in sense of starting and ending the scan but no devices are shown.

PS: I have kept 3 devices near the mobile where I am doing USB debugging of the app.

Output of ionic info

```
Ionic:

   ionic (Ionic CLI)  : 4.10.1 (/usr/lib/node_modules/ionic)
   Ionic Framework    : ionic-angular 3.9.3
   @ionic/app-scripts : 3.2.1

Cordova:

   cordova (Cordova CLI) : 8.1.2 (cordova-lib@8.1.1)
   Cordova Platforms     : android 7.1.4
   Cordova Plugins       : cordova-plugin-ionic-keyboard 2.1.3, cordova-plugin-ionic-webview 3.1.2, (and 5 other plugins)

System:

   Android SDK Tools : 26.1.1 (/home/akash/Android)
   NodeJS            : v10.15.0 (/usr/bin/node)
   npm               : 6.7.0
   OS                : Linux 4.18
```

Looking for some solution or any other alternative to this.
My aim is to establish connection between two devices nearby and share some information between them most probably without using internet.
Thank you

randdusing, 2019-02-27:
The other Android devices need to be advertising as BLE devices. They won't do this by default. An application needs to be advertising its services.

AkashRajpurohit, 2019-02-27:
So I tried this now.

Added a startAdvertising function which is called on a button click event.

I have started the app on two android devices.
From one I started advertising using the demo code from here - https://github.com/randdusing/cordova-plugin-bluetoothle#startadvertising 
Passing the same params.

From another device, I'm starting the scan.
Still no luck. Can't find the other mobile which is emitting the advertising event.

Also I tried the isAdvertising() method. It returns undefined in every case.

But start and stop Advertising returns back a status properly.

Can you help me debug this?

randdusing, 2019-02-28:
Try using the example in the angular wrapper. If you install the sample app on two devices, they will communicate.

ragcsalo, 2020-03-11:
Connecting one or more Android central devices to an Android peripheral device works for me. :-) 

However, notifying subscribed devices seems to stop unexpectadly after some time...The notify command returns a success, but still, nothing received on the subscribed devices. This seems to happen randomly, no exact timeframe... sometimes after 2 minutes, sometimes after 30 seconds.

Very interesting, because all subscribed iOS devices are still getting notified for an unlimited time... only Android to Android links seem to get broken randomly. 🤔

clarklight, 2021-09-13:
> Try using the example in the angular wrapper. If you install the sample app on two devices, they will communicate.

Is it possible to pass data back and forth between the 2 android devices with this plugin? If i can manage to connect the 2 devices?

asif166037, 2025-10-17:
@AkashRajpurohit , I am also facing this issue when we using two android device , only getting this 

{
   "status": "scanStarted"
   }