randdusing/cordova-plugin-bluetoothle #530: No callback when there is a connection in peripheral(iOS)

nguyenvt2, 2019-03-14:
I'm using initializePerheral() to init bluetooth for peripheral(iOS), addService() and startAdvertising(), but when there is a connection, don't have any callback! Only callback in case: enabled, disabled, subscribed, unsubscribed!

randdusing, 2019-03-15:
Try running the example in the angular wrapper

nguyenvt2, 2019-03-15:
> Try running the example in the angular wrapper

Thank for reply! I tried to run the example but ionic v1 is too old to build ios application(can't use ionic v4 to running it). Can you help me review your guide? Thanks!

randdusing, 2019-03-15:
I've started using Angular (2+) and observables much more often at my day job. When I originally created the plugin and AngularJS wrapper a few years ago, I knew nothing about Angular (2+) nor rxjs. I can't do anything in the near future due to some important real life commitments, but perhaps I can create  a new example based on Ionic 4.

randdusing, 2019-03-15:
I still recommend checking out the code in the example and make sure you're following the same pattern.

nguyenvt2, 2019-03-15:
Thanks for your answers! I am developing it on ionic 4, I tried to learn a lot but failed, very much looking forward to your example on ionic 4!

tvalentino, 2020-01-15:
I too am seeing the same issue.  Running this in a Cordova (phonegap) implementation.  Structured the code to be similar to the angular example, but still only get the 4 callbacks -- enabled, disabled, subscribed, unsubscribed.

Sebartus, 2022-01-26:
> I too am seeing the same issue. Running this in a Cordova (phonegap) implementation. Structured the code to be similar to the angular example, but still only get the 4 callbacks -- enabled, disabled, subscribed, unsubscribed.

@tvalentino did you find any solution ?

@randdusing after bluetoothle.initializePeripheral(success, error, { request: true }); and connect to BLE device readRequested status never fired on iOS but Android works perfect.