randdusing/cordova-plugin-bluetoothle #531: Peripheral mode : Unable to add descriptor to characteristic

MajidAliKhanQuaid, 2019-03-15:
How can i define a descriptor in characteristics ? Couldn't find a way in

MajidAliKhanQuaid, 2019-03-15:
var params = {
  service: "1234",
  characteristics: [
    {
      uuid: "ABCD",
      permissions: {
        read: true,
        write: true,
        //readEncryptionRequired: true,
        //writeEncryptionRequired: true,
      },
      properties : {
        read: true,
        writeWithoutResponse: true,
        write: true,
        notify: true,
        indicate: true,
        //authenticatedSignedWrites: true,
        //notifyEncryptionRequired: true,
        //indicateEncryptionRequired: true,
      }
    }
  ]
};

How would i set default value for the descriptor

MajidAliKhanQuaid, 2019-03-15:
And, How can i get that address parameter ?

MajidAliKhanQuaid, 2019-03-15:
notify results in error, *address is not defined*

randdusing, 2019-03-15:
It's not possible to set a default value for the descriptor with the current code.

The #notify() documentation is out of date. An address needs to be passed in. The address can be found from the status updates of #initializePeripheral()

MajidAliKhanQuaid, 2019-03-15:
thanks @randdusing for the reply. Nice contribution 👍

closed 2019-03-15 by MajidAliKhanQuaid