randdusing/cordova-plugin-bluetoothle #534: retrieveConnected success callback always returns an empty parameter

jaksawesome10, 2019-04-19:
Whenever I use retrieveConnected, the success callback has a parameter called devices, which is always equal to "", every time the function is called, even when a bluetooth device is connected to the device running my code.  I have tested and the app Im creating is initialized, blueooth is enabled, and all permissions have been granted

jaksawesome10, 2019-04-19:
By the way this is my code,

```
bluetoothle.retrieveConnected(
									
function(devices){
										
	//success
	if(devices.length == 0){
		
		console.log("No devices connected");
		
	}
										
	else{
											
		for(var i = 0; i < devices.length; i++){
												
			console.log(status[i]);
												
		}
											
	}
										
	}, function(e){
										
		//error
										
		console.log("Error connecting : " + e);
											
	}, function(param){
										
	//parameters
										
	console.log(param);
									
});
```

randdusing, 2019-05-12:
I believe connected only returns devices that are "paired"/saved in the Bluetooth settings device list.