randdusing/cordova-plugin-bluetoothle #537: Renaming on iOS

Thyrion33, 2019-05-20:
Hello,
I use your plugin for an App intended for iOS and Android.
The device is renamed with a special characteristic.
The renaming works fine on both Android and iOS. The problem is that when I scan after renaming the device name is not changed on iOS, i have to wait some time. On Android it's work fine.
Is that some sort of cache ? Any way to fix it ?
Thanks in advance !

aroncal, 2019-05-21:
I had this same problem few years ago when I designed my App. Finally, my solution was to do all this after Bluetooth name change:

* disconnect
* close connection
* start scan
* connect
* discover

with small sleeps between the commands

Thyrion33, 2019-05-27:
Tanks for your answer aroncal.

Hm i was already doing the same steps, but without sleeps between disconnect and close connection, but adding one doesn't change anything.
How many time is your delay between each commands ?

aroncal, 2019-05-27:
disconnect -> (100ms) -> close conn -> (2000ms) -> start scan

Thyrion33, 2019-05-29:
That's doesn't fix the problem for me.