randdusing/cordova-plugin-bluetoothle #538: Writing number

untilbit, 2019-05-31:
How to write numbers as encoded strings?

I tried several times but it's not working.

randdusing, 2019-06-01:
It depends on the type of number, which depends on what the peripheral expects.

untilbit, 2019-06-01:
@randdusing I want to write uInt16, uInt32, boolean and also strings.

untilbit, 2019-06-03:
Tried several ways with no success. Here is an example of what I'm trying to write:

```
  write(deviceId, service, characteristic, value) {
    const bytes = this.bluetoothle.stringToBytes(value);
    const encodedString = this.bluetoothle.bytesToEncodedString(bytes);

    this.bluetoothle.write({
      address: deviceId,
      service: service,
      characteristic: characteristic,
      value: encodedString
    }).then(
      (result) => alert('OK: ' + result.status),
      (error) => alert('Error: ' + JSON.stringify(error))
    );
  }
```

untilbit, 2019-06-06:
@randdusing Could you help? Thanks.

randdusing, 2019-06-06:
It just depends on how those types are stored at a byte level. Here's an example of long to byte array: https://stackoverflow.com/questions/8482309/converting-javascript-integer-to-byte-array-and-back#12965194

A boolean would probably just be:
```
var trueBytes = [1];
var falseBytes = [0];
```

untilbit, 2019-06-10:
When I try to write a number directly like in the example above to a characeristic I dont get success nor error callback.

But if I read a descriptor and writing a number like in the example above to a characeristic works only on Android 8 and 9.

- Android 9: Write OK.
- Android 8: Write OK.
- Android 7: **Unable to write on return** error.
- Android 6: **Unable to write on return** error.
- Android 5: **Unable to write on return** error.

randdusing, 2019-07-15:
Do other writes work on 5-7?