randdusing/cordova-plugin-bluetoothle #541: Discover hangs indefinitely

NunoSav, 2019-06-06:
I'm building a ionic+capacitor app and using this plugin to connect to BLE devices.
**Sometimes** discover method hangs, no response or error thrown. I get the following on debug window:

```
V/Capacitor/Plugin: To native (Cordova plugin): callbackId: BluetoothLePlugin163978616, service: BluetoothLePlugin, action: discover, actionArgs: [{"address":"D0:CF:5E:B2:46:20","clearCache":true}]
D/BluetoothGatt: refresh() - device: D0:CF:5E:B2:46:20
D/BluetoothGatt: discoverServices() - device: D0:CF:5E:B2:46:20
D/BluetoothGatt: onClientConnParamsChanged() - Device=D0:CF:5E:B2:46:20 interval=6 status=0
D/BluetoothGatt: onClientConnParamsChanged() - Device=D0:CF:5E:B2:46:20 interval=39 status=0
```

A similar issue has been reported [here](https://stackoverflow.com/questions/43925466/bluetoothgatt-onclientconnparamschanged).

gianted, 2019-06-19:
Hi, I've been using this plugin for a long time but since Android 9 I'm not able to discover any peripheral device anymore. Issue observed with cordova-plugin-bluetoothle v4.5.5. The peripheral works in non-paired way by design; startScan uses this parms:

`var params = {
        "services": service,
        "allowDuplicates": false,
        "scanMode": bluetoothle.SCAN_MODE_LOW_LATENCY,
        "matchMode": bluetoothle.MATCH_MODE_AGGRESSIVE,
        "matchNum": bluetoothle.MATCH_NUM_MAX_ADVERTISEMENT,
        "callbackType": bluetoothle.CALLBACK_TYPE_ALL_MATCHES
    };`

There's some issues on API level 28? Please advise if some debug log is requested and in which way.

randdusing, 2019-07-15:
@gianted Did you try using the default params, or changing params one at a time to see if they have any effect?

gianted, 2019-07-15:
> @gianted Did you try using the default params, or changing params one at a time to see if they have any effect?

Yes @randdusing I tried also other options, but without success.

Meanwhile, I set the "targedSdkVersion" to API level 26, instead 28 into "config.xml"

`<preference name="android-targetSdkVersion" value="26" />`

and the plugin seems to work again (even if it seems less efficient in terms of scanning and connecting features: I have timeout issues more frequently than before).

NunoSav, 2019-07-15:
Forgot to update this issue. The problem i had with `discover` method is solvable by turning bluetooth off and on (i believe you also need to unbond).

Not sure where the fault lies here but it seems that it is on the Android side (API 23).
@randdusing feel free to close this one.

NunoSav, 2021-04-22:
In case someone else has this issue just implement a timeout that disconnects+close connection and tries connect+discover again.

closed 2021-04-22 by NunoSav