randdusing/cordova-plugin-bluetoothle #542: Unable to write on android

SandroMoretti, 2019-07-04:
Hi guys, I'm follwing the documentation.
Work perfectly on iOS. But on android I received the following error on write:

```
address: "A4:CF:12:25:D1:B2"
characteristic: "110F6A8D-503F-4F17-A9E3-F3169553A803"
error: "write"
message: "Unable to write"
name: "Ovo Fácil"
service: "110F6A8D-503F-4F17-A9E3-F3169553A803"
```

I'm using the follow commands:
```
bluetoothle.initialize();
...startScan();
...stopScan();
...connect(); <- passing correct address
...discover(); <- passing correct addres
bluetoothle.write(function(s){console.log(s);}, function(e){console.log(e);}, {
address: "A4:CF:12:25:D1:B2",
service: "110F6A8D-503F-4F17-A9E3-F3169553A803",
characteristic: "110F6A8D-503F-4F17-A9E3-F3169553A803",
value: encodedString
});
```

Discover is returning this:
```
characteristics: Array(2)
0: {uuid: "110F6A8D-503F-4F17-A9E3-F3169553A803", properties: {…}, permissions: {…}, descriptors: Array(1)}
1:
descriptors: []
permissions: {}
properties: {write: true}
uuid: "110F6A8D-503F-4F17-A9E3-F3169553A803"
__proto__: Object
length: 2
__proto__: Array(0)
uuid: "110F6A8D-503F-4F17-A9E3-F3169553A803"
```

Someone see some error in my life cycle? Some mistake? Thanks.

<b>Cordova version:</b> 8
<b>Android SDK level:</b> 27
<b>Smartphone: </b> Xiaomi A2 Lite (Android 9)

Ariaal, 2019-07-05:
Hello, may I ask, how did you write on ios? My program can't run Bluetooth directly.
first I add cordova-plugin-bluetooth cordova-plugin-background-mode-bluetooth-central,and then i write this in my code: 
 bluetoothle.initialize(
        {
          "request": true
        },
        function(obj) {
          console.log(JSON.stringify(obj));
          console.log('obj');
          bluetoothle.startScan(startScanSuccess, function(obj){
            alert("扫描失败");
            alert(JSON.stringify(obj));
          }, {
            services: []
          });
          var startScanSuccess = function(device) {
            if (device.status == "scanStarted") {
              console.log("设备开始扫描");
            } else if (device.status == "scanResult") {
              if (device.name) {
                if (
                  device.name.indexOf("BPAP") >= 0 ||
                  device.name.indexOf("CPAP") >= 0
                ) {
                  that.devicelist.push(device);
                }
              }
            }
          };
xcode error:
2019-07-05 13:32:10.549831+0800 imatrixh5app[389:99471] [CoreBluetooth] API MISUSE: <CBCentralManager: 0x281912ac0> has no restore identifier but the delegate implements the centralManager:willRestoreState: method. Restoring will not be supported
2019-07-05 13:32:10.557407+0800 imatrixh5app[389:99471] {"withPromises":{}}
2019-07-05 13:32:10.630579+0800 imatrixh5app[389:99471] Error in Success callbackId: BluetoothLePlugin1791913246 : TypeError: callback.success.apply is not a function. (In 'callback.success.apply(null, args)', 'callback.success.apply' is undefined)
2019-07-05 13:32:10.630726+0800 imatrixh5app[389:99471] callbackFromNative@file:///var/containers/Bundle/Application/B0AA5F40-B857-4F40-A087-68311AC9E935/imatrixh5app.app/www/cordova.js:288:63
nc2@file:///var/containers/Bundle/Application/B0AA5F40-B857-4F40-A087-68311AC9E935/imatrixh5app.app/www/cordova.js:1039:39
2019-07-05 13:32:21.590013+0800 imatrixh5app[389:99471] {"withPromises":{}}
2019-07-05 13:32:21.605975+0800 imatrixh5app[389:99471] Error in Success callbackId: BluetoothLePlugin1791913247 : TypeError: callback.success.apply is not a function. (In 'callback.success.apply(null, args)', 'callback.success.apply' is undefined)

Do you have some idea for me?
thanks!

SandroMoretti, 2019-07-05:
Hi @Ariaal , I don't know if have some better way, but I made with this:

https://gist.github.com/SandroMoretti/184037760a81d347067ab74db0f03c2e

Ariaal, 2019-07-05:
Thank you so much!

BobBDE, 2019-10-08:
Hi @SandroMoretti, I have the same exact problem, dit you find the solution to this ?

SandroMoretti, 2019-10-11:
Hi @BobBDE , sadly but I don't find any solution.
I'll try to use another plugin on android.

cordova-plugin-bluetoothle to iOS and cordova-plugin-blueooth-serial to android.

randdusing, 2019-10-12:
I believe cordova-plugin-bluetooth-serial is only for normal Bluetooth, not BLE.