randdusing/cordova-plugin-bluetoothle #543: Error in Success callbackId: BluetoothLePlugin252653612 : TypeError: callback.success.apply is not a function.

Ariaal, 2019-07-05:
am using vue and cordova as a hybrid APP to implement Bluetooth data transfer. First I added plugin cordova-plugin-bletoothle and add cordova-plugin-background-mode-bluetooth-central . Then I used the plugin API directly in my project. Here is my usage:

bluetoothle.startScan(startScanSuccess, function(obj){
            alert("scan wrong");
            alert(JSON.stringify(obj));
          }, {
            services: []
          });
          var startScanSuccess = function(device) {
            if (device.status == "scanStarted") {
            } else if (device.status == "scanResult") {
              console.log(device);
              
              if (device.name) {
                if (
                  device.name.indexOf("BPAP") >= 0 ||
                  device.name.indexOf("CPAP") >= 0
                ) {
                  that.devicelist.push(device);
                }
              }
            }
          };

xcode: 
init bluetooth
2019-07-05 15:53:08.049122+0800 imatrixh5app[479:118710] [CoreBluetooth] API MISUSE: <CBCentralManager: 0x2827b7880> has no restore identifier but the delegate implements the centralManager:willRestoreState: method. Restoring will not be supported
2019-07-05 15:53:08.050630+0800 imatrixh5app[479:118710] ERROR: {"line":165,"column":20}
2019-07-05 15:53:08.076212+0800 imatrixh5app[479:118710] Error in Success callbackId: BluetoothLePlugin252653612 : TypeError: callback.success.apply is not a function. (In 'callback.success.apply(null, args)', 'callback.success.apply' is undefined)
2019-07-05 15:53:08.076375+0800 imatrixh5app[479:118710] callbackFromNative@file:///var/containers/Bundle/Application/A73C0CF5-7713-4400-8819-816412A8EB49/imatrixh5app.app/www/cordova.js:288:63
nc2@file:///var/containers/Bundle/Application/A73C0CF5-7713-4400-8819-816412A8EB49/imatrixh5app.app/www/cordova.js:1039:39

anyone can help? I use javascript

venatiodecorus, 2019-07-05:
your params need to be the last argument to your initialize function

closed 2019-07-05 by randdusing

Ariaal, 2019-07-09:
@venatiodecorus  thanks so much! I can use it now!!