randdusing/cordova-plugin-bluetoothle #544: Cordova 9 + Angular 8 can't find name bluetoothle

thebigandrei, 2019-07-09:
Hello !

I would like to use this plugin in a cordova 9 + angular 8 app, is it possible?

I'm kind of new with cordova, and i don't really understand how to use the plugin in my angular code.

I installed the plugin correctly but can't access the plugin from my angular service (can't find name bluetoothle).

Let me know all informations you want me to give about my project.

Thanks.

venatiodecorus, 2019-07-10:
you can access the plugin with `(<any>window).bluetoothle`

thebigandrei, 2019-07-10:
This is the code in my angular service.

`(<any>window).bluetoothle.isEnabled(success=> {
      console.log(success);
    });`

This returns a new error : "Class not found".

Thanks for the effort =)

closed 2019-07-10 by thebigandrei