randdusing/cordova-plugin-bluetoothle #545: calbackMessage: Service not found

Ariaal, 2019-07-11:
I get serveUUID on LightBlue. My serviceUUID is 128 bit,but I notify that uuid is 16bit,how can i get uuid ? I use bluetooth.discover to get it,but fail.

mycode:  
handle(item) {
      bluetoothle.connect(
        function(res) {
          if (res.status === "connected") {
            console.log(res);
            
            // this.$router.push({ name: "connect", query: { id: item.address } });
            bluetoothle.discover(
              function(res) {
                console.log("kill" + JSON.stringify(res));
              },
              function(res) {
                console.log("fail" + JSON.stringify(res));
              },
              {
                address: item.address
              }
            );
          } else if (res.status === "disconnected") {
            alert(res);
          }
        }.bind(this),
        function(failCallback) {
          alert(failCallback);
        },
        {
          address: item.address
        }
      );

dockinkong, 2019-07-12:
Same problem on Android... Discover called in connect succes callback fails with error "Device isn't connected".  Discover seemes to be longer than the time connection could be established..

dockinkong, 2019-07-12:
Is discover absolutely needed before sending write action?

randdusing, 2019-07-15:
Yes, the characteristic must be discovered before using it.

dockinkong, 2019-07-15:
Thanks for your reply... do you know why is device disconnected so quickly? I could not perform discover before beeing disconnected...