randdusing/cordova-plugin-bluetoothle #547: allowDuplicates does not even filter out same device.

biswas123, 2019-08-04:
The field allowDuplicates: false is not  filtering out same detected BT device.

randdusing, 2019-08-04:
Please provide more details: operating system, phone, example code, etc

biswas123, 2019-08-05:
- oneplus 6 running Android P
- code is same from the docs with allowDuplicates set to false.

I am from my phone so cannot share the code sorry.

randdusing, 2019-08-09:
Please share the code when you can. It could be a side effect of the phone or Android version

adelin-mcbsoft, 2021-01-16:
I confirm this is happening on Android (while on iOS works correctly).

Phone: *Huawei Y6 2019 (MRD-LX1)*
Android Version: *9*

Basic code used:
```
bluetoothle.startScan(function(s) {
    console.log(['SUCCESS', s]);
}, function(e) {
    console.log(['ERROR', e]);
}, {
    "allowDuplicates": false
});
```
Results in endless duplicates of the same device...

adelin-mcbsoft, 2021-01-16:
As per my basic code understanding, there isn't any implementation at all regarding `allowDuplicates` in the Android's [BluetoothLePlugin.java](https://github.com/randdusing/cordova-plugin-bluetoothle/blob/master/src/android/BluetoothLePlugin.java) file _(meaning a quick search for this tag finds nothing, while it can be found in the iOS's [BluetoothLePlugin.m](https://github.com/randdusing/cordova-plugin-bluetoothle/blob/master/src/ios/BluetoothLePlugin.m) file_.

So I guess it was just forgotten... wish it'd be a quick feature to implement.