randdusing/cordova-plugin-bluetoothle #548: Arguments error when providing no arguments for connect

gokujy, 2019-08-08:
Hi randdusing

First of all, this plugin sounds very promising for me- I'm trying to get my app to include BLE for locational data on Android and iOS, so thanks very much for your efforts.

At the moment I'm working on the Android version.

But as I've tried to scan and connect for BLE signals, I've provided device address as params, and I'm getting "error = {error: "arguments", message: "Argument object not found"}".

What to do, please help

randdusing, 2019-08-09:
Ensure the method is called with params ```bluetoothle.startScan(startScanSuccess, startScanError, params);``` Check out the example in the Angular wrapper

gokujy, 2019-08-12:
I am using like this:

`this.ble.connect({ "address": address }).subscribe((data) => {
 this.showToast(JSON.stringify(data));
 }, (error) => {
 this.showError("device not connect " + JSON.stringify(error));
 })`

venatiodecorus, 2019-08-13:
are you calling `startScan` somewhere? where are you getting `address` from?

geocory, 2019-09-30:
@gokujy I followed the link to your demo code, and on line 159 of home.page.ts you are not providing an object. The code sample you provided on Aug 11 is not what is in home.page.ts. 

Instead of having this as you claim:
`this.ble.connect({ "address": address })`

The code in the repo has:
`this.bluetoothle.connect(device.address)`

You need to change the code on line 159 to be:
`this.bluetoothle.connect( {"address": device.address} )`