randdusing/cordova-plugin-bluetoothle #550: can't set android device name  when advertizing

gamjaradio, 2019-09-02:
I can't change device name whne advertizing. It always shows my phone name.
my android is version 9, and 'cordova-plugin-bluetoothle": "^4.5.5".

var params = {
                services: [LIGHTBULB_SERVICE],
                service: LIGHTBULB_SERVICE,
                name: "hi",
                mode: "balanced",
                connectable: true,
                timeout: 20000,
                powerLevel: "high",
                includeDeviceName: true
            };
            
var v = await this.bluetoothle.startAdvertising(params);

this is my source,
is there a problem??

randdusing, 2019-09-02:
The Android code doesn't use the name parameter. It just uses the includeDeviceName boolean. Readme most likely needs to be updated.

gamjaradio, 2019-09-02:
thanks the comment :)
Then one more question, 
is it not possible to change the bluetooth name on Android?

randdusing, 2019-09-28:
Not that I know of...unless you just update your device's actual name.

ragcsalo, 2020-03-09:
There is a solution: in BluetoothLePlugin.java search this line
`BluetoothLeAdvertiser advertiser = bluetoothAdapter.getBluetoothLeAdvertiser();`

and insert these 2 lines below:
```
String advertisedName = obj.optString("name", "BTLE-default-name");
bluetoothAdapter.setName(advertisedName);
```

gamjaradio, 2020-03-10:
It works 👍. Thanks a lot !!

closed 2020-03-10 by gamjaradio