randdusing/cordova-plugin-bluetoothle #551: I can only send data from ESP32 to Android Phone once.

erhankav, 2019-09-02:
Hi Everybody,

On the Android phone, there are no problems in finding addresses, identifying services, determining characteristics. However, ESP32 is constantly sending data and notified changed data, showing only one of these data on the screen and the application disconnects.

The line of code says "Once we're done reading all the values, disconnect". However, as I mentioned, esp32 continues to send data.

Does anyone have an idea about it

Thanks.

Erhan

randdusing, 2019-09-02:
Is that code from your application layer? You'll need to post example code

erhankav, 2019-09-03:
**Hi, Rand

You can see the ESP32 sample code in the following lines**

   Create a BLE server that, once we receive a connection, will send periodic notifications.
   The service advertises itself as: 4fafc201-1fb5-459e-8fcc-c5c9c331914b
   And has a characteristic of: beb5483e-36e1-4688-b7f5-ea07361b26a8

   The design of creating the BLE server is:
   1. Create a BLE Server
   2. Create a BLE Service
   3. Create a BLE Characteristic on the Service
   4. Create a BLE Descriptor on the characteristic
   5. Start the service.
   6. Start advertising.

   A connect hander associated with the server starts a background task that performs notification
   every couple of seconds.
*/
#include <BLEDevice.h>
#include <BLEServer.h>
#include <BLEUtils.h>
#include <BLE2902.h>

BLEServer* pServer = NULL;
BLECharacteristic* pCharacteristic = NULL;
bool deviceConnected = false;
bool oldDeviceConnected = false;
uint32_t value = 0;

#define SERVICE_UUID        "4fafc201-1fb5-459e-8fcc-c5c9c331914b"
#define CHARACTERISTIC_UUID "beb5483e-36e1-4688-b7f5-ea07361b26a8"

class MyServerCallbacks: public BLEServerCallbacks {
    void onConnect(BLEServer* pServer) {
      deviceConnected = true;
    };

    void onDisconnect(BLEServer* pServer) {
      deviceConnected = false;
    }
};

void setup() {
  Serial.begin(115200);

  // Create the BLE Device
  BLEDevice::init("ESP32");

  // Create the BLE Server
  pServer = BLEDevice::createServer();
  pServer->setCallbacks(new MyServerCallbacks());

  // Create the BLE Service
  BLEService *pService = pServer->createService(SERVICE_UUID);

  // Create a BLE Characteristic
  pCharacteristic = pService->createCharacteristic(
                      CHARACTERISTIC_UUID,
                      BLECharacteristic::PROPERTY_READ   |
                      BLECharacteristic::PROPERTY_WRITE  |
                      BLECharacteristic::PROPERTY_NOTIFY |
                      BLECharacteristic::PROPERTY_INDICATE
                    );

  // https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.descriptor.gatt.client_characteristic_configuration.xml
  // Create a BLE Descriptor
  pCharacteristic->addDescriptor(new BLE2902());

  // Start the service
  pService->start();

  // Start advertising
  BLEAdvertising *pAdvertising = BLEDevice::getAdvertising();
  pAdvertising->addServiceUUID(SERVICE_UUID);
  pAdvertising->setScanResponse(false);
  pAdvertising->setMinPreferred(0x0);  // set value to 0x00 to not advertise this parameter
  BLEDevice::startAdvertising();
  Serial.println("Waiting a client connection to notify...");
}

void loop() {
    // notify changed value
    if (deviceConnected) {
        pCharacteristic->setValue((uint8_t*)&value, 4);
        pCharacteristic->notify();
        value++;
        delay(3); // bluetooth stack will go into congestion, if too many packets are sent, in 6 hours test i was able to go as low as 3ms
    }
    // disconnecting
    if (!deviceConnected && oldDeviceConnected) {
        delay(500); // give the bluetooth stack the chance to get things ready
        pServer->startAdvertising(); // restart advertising
        Serial.println("start advertising");
        oldDeviceConnected = deviceConnected;
    }
    // connecting
    if (deviceConnected && !oldDeviceConnected) {
        // do stuff here on connecting
        oldDeviceConnected = deviceConnected;
    }
}

**Cordova codes are the same as your sample codes. I replaced the "disconnect" functions with the "close" functions. But again I get the same result. the application reads once, but disconnects without reading, while the data sent by ESP32 continues.

I was wondering if there was a problem with the ESP32 code. But I downloaded the Nordic nRF Connect application to the phone and connected to ESP32. I receive data regularly.

I would like your support.

Thanks.**

randdusing, 2019-09-28:
Sorry, I was expecting application code from the JavaScript side. I don't know have any knowledge of ESP32 code.

chegewara, 2019-11-16:
Im building ionic app that will work with esp32 and i can confirm that all works so far. 
@erhankav My suggestion is to increase delay after notify, because 3 ms may cause congestion. Start with 100 or 1000 ms and then you can decrease it.

adrian-fratila, 2019-12-22:
Hi guys,

I am a newbie in Ionic, and I try to get working a BLE peripheral in Ionic 3. For some reason the cordova bluetoothle plugin returns as not found when I try to build, although it appears as installed correctly when I try to reinstall it. Nevertheless somehow I got the @ionic-native/bluetooth-le to build the code.

I followed the steps of initialise, initializePeripheral, addService, startAdvertising and they return like everything as expected (ending up with AdvertisingStarted), but for some reason the device doesn't appear when I scan from another smartphone.

Also I could not figure out how to assign the callback function which is called when the device gets connected to, or receives something from the other side / the client.

I would highly appreciate if you could point me to an example code which implements the peripheral functionality in Ionic 3, as I have been struggling already a couple of days on this issue..

Thank you,
Adrian