randdusing/cordova-plugin-bluetoothle #552: how to add custom data on advertizing

gamjaradio, 2019-09-07:
I want to set some information to Advertisement without connection read/write.
There is some for it in android bluetooth library, but I can not find here in cordova plugin.

randdusing, 2019-09-28:
Looks like the readme is out of data. You want to use #startAdvertising() with manufacturerSpecificData
https://github.com/randdusing/cordova-plugin-bluetoothle/blob/master/src/android/BluetoothLePlugin.java#L738

gamjaradio, 2019-09-28:
thanks a lot :). I cat make it work with manufacturerSpecificData.

> Looks like the readme is out of data. You want to use #startAdvertising() with manufacturerSpecificData
> https://github.com/randdusing/cordova-plugin-bluetoothle/blob/master/src/android/BluetoothLePlugin.java#L738

closed 2019-09-28 by gamjaradio