randdusing/cordova-plugin-bluetoothle #556: Infinite loop when adding descriptors

PacificNorthwest, 2019-10-29:
Hey, just wanted to say I appreciate the work you're doing on your contribution.
Thought there's a pretty severe problem that causes an infinite loop whenever one tries to add a characteristic descriptor on Android.

Loop parameters seems to be off here:
https://github.com/randdusing/cordova-plugin-bluetoothle/blob/2c35583c073e9add0babd5b7e9302af47ed7bb85/src/android/BluetoothLePlugin.java#L563

And also here:
https://github.com/randdusing/cordova-plugin-bluetoothle/blob/2c35583c073e9add0babd5b7e9302af47ed7bb85/src/android/BluetoothLePlugin.java#L567

Figured you might want to check into it. Thanks!

randdusing, 2019-11-10:
Oops, that's pretty bad. I'm surprised this wasn't found earlier!

randdusing, 2019-11-10:
This has been fixed in v4.5.6

closed 2019-11-10 by randdusing