randdusing/cordova-plugin-bluetoothle #558: Cannot Connect on IOS 13

agileMikeDev, 2019-11-15:
Hi.

Im am runniing an Ionic (V3) Cordova (V8) Application.
Everythis was running fine so far.

Since IOS Version 12 it seems connecting to a BLE Device became really slow.
Before IOS12 the connection was establishes in a few ms. Since V12 the connections to our Devices took about5 to 7 seconds. 

Now on Devices with IOS 13 co connection can be established at all!!!!
Wether the success nor the error callback is called. The connection always runs into timeout.

Has anyone an idea what changed here in the ios bluetooth core that could explain this behaviour?

m4tta, 2019-11-16:
Are you using the latest release? I have no connection problems to a brand new iPhone 11 running iOS 13.2. The only thing that changed on Apples side of things is a new key `NSBluetoothAlwaysUsageDescription` that needs to be added to your info.plist

https://developer.apple.com/documentation/bundleresources/information_property_list/nsbluetoothperipheralusagedescription

agileMikeDev, 2019-11-19:
Hi there.

After some more investigation we found out, that the Bluetooth Devices, we could not connect, did not implement all advertisement packages. In our special case the PHY information was not send. Before IOS13 this was no Problem, Since the new IOS Version Applle seems to be more strict and does not establis a connection. After the missing PHY information was implemented in the BLE Device stack, everithing is running fine again... 

Thanx mike.

closed 2019-11-19 by agileMikeDev