randdusing/cordova-plugin-bluetoothle #559: bug in retrieveConnected

chegewara, 2019-11-16:
When i print data from retrieve connected i am getting only 2 devices, but i know i have 3 devices bonded (confirmed with nRF connect), also `isBonded` returns true with my TV address, which is not included in `retrieveConnected`:
```
[console.log]: [{"address":"7A:68:61:3F:C8:D1","name":"Galaxy S5"},{"address":"2C:BA:BA:2F:91:3B","name":"Gear VR Controller(913B)"}]
```

ionic + angular on Samsung S9+, android 9.

asoap, 2019-11-22:
I am having a similar issue on Android.  I bought a smart watch to interface with the heart rate monitor.  The watch is connected to the phone but when I call retrieveConnected() I only get devices that my specific app has connected to.  It doesn't retrieve the watch.

randdusing, 2019-12-16:
@asoap Thanks for taking a closer look. This should be fairly simple to fix, so I'll try to raise a PR in the next week or so. But PRs are always welcome :)

asoap, 2019-12-16:
I can make a pull request and do the change.  I'm just not sure if it should be changed?  Like I don't know enough about BLE to know if DEVICE_TYPE_DUAL should be talking also?  Or if there is some limit that is added because of it?

randdusing, 2019-12-24:
I don't think there's any issue updating it to DEVICE_TYPE_DUAL.

https://stackoverflow.com/questions/54922664/android-bluetoothdevice-type-dual

ragcsalo, 2020-09-24:
bluetoothle.retrieveConnected doesn't return any paired devices on iOS 14 (iPhone X)

czzplnm, 2020-12-23:
bluetoothle.retrieveConnected is returning a blank array to me for the success callback.

MojioMS, 2021-04-06:
I confirm obtaining an empty array in the `retrieveConnected` success response, although being bonded with a device. Tested on three Android devices.
Another bond() attempt on the already bonded address returns a correct `Device already bonded` error message and `isBonded` with the address of the bonded device returns true.

PR #683 (2021-04-07): Update BluetoothLePlugin.java
fixes #559
closed 2021-04-07 by randdusing