randdusing/cordova-plugin-bluetoothle #562: Does this plugin support the BLE 4.0 Mesh protocol?

Lizard-King101, 2019-12-03:
was looking for a plugin that can perform Bluetooth 4.0 Mesh functionality. would Love to see this functionality be taken advantage of and would really love to make an application on top of this functionality.

randdusing, 2019-12-16:
I'm not too familiar with Bluetooth mesh, but the plugin supports almost all the Bluetooth LE related API calls available for Android and iOS. What do API do you need supported specifically?

Lizard-King101, 2019-12-16:
Its called BLE Mesh API I think it needs another library included from [NordicSemiconductors](https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library) and [iOS](https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library) To be fair this might be more than you might want to do but I would absolutely LOVE if you could implement this API.