randdusing/cordova-plugin-bluetoothle #565: iOS 13.3 - Unexpected Error in plugin

jkeczan, 2019-12-18:
Since upgrading to iOS 13.3, I am getting a consistent process error from my app when installed on my phone. 

2019-12-18 10:05:02.758020-0500 App[1786:108577] [Process] kill() returned unexpected error 1

This returns consistently and goes away and if I don't initialize bluetooth so am thinking there is something going on there.

ndrake, 2019-12-18:
Per [this](https://stackoverflow.com/questions/58321114/why-i-get-the-console-warning-process-kill-returned-unexpected-error-1-when), it seems to be a Webkit bug

jkeczan, 2019-12-18:
That is certainly similiar. Odd thing is that it only started post update to 13.3 and stops if I don't init the Ble plugin.

randdusing, 2019-12-24:
Does it affect any of the Bluetooth functionality? If it does, can you post any lower level BLE logs?