randdusing/cordova-plugin-bluetoothle #570: initialize does not trigger status changes

pallinp, 2020-01-17:
Hi,

I've got a little problem with the "initialize" method.

The first subscription trigger is working and I've got the Bluetooth status. When I turn on or turn off the Bluetooth later, the subscription is not triggered anymore.

Am I missing something ?

My plan was to use the subscription to manage properly my resources when the Bluetooth is turned off wildly.

Here is my code:

```ts
this.ble.initialize({
  'request': true,
  'statusReceiver': false,
  'restoreKey': 'sandbox'
}).subscribe((status) => {
  console.log(status);
});
```

I'm using Ionic 4 and the version 4.5.6 of the Bluetooth plugin.

I've got the same behavior on a Samsung Galaxy tab 3 running Android 9 and on an Xperia G3212 running Android 8.

Thanks in advance

Regards

JesperBalslev, 2020-02-03:
Just to add on this, It works as expected on iOS but not on Android.

JesperBalslev, 2020-02-03:
I fixed the issue in the plugin (Android) - but before i make a PR - i will dig deeper and see if there are other  issues i can fix...

randdusing, 2020-02-12:
statusReceiver => true doesn't help with this?

pallinp, 2020-02-12:
@randdusing: It does ! I misunderstood the documentation. Thanks.

closed 2020-02-12 by pallinp

OJ7, 2020-02-25:
> statusReceiver => true doesn't help with this?

It seems like either the code or documentation needs updating.
On iOS, it triggers status changes by default. (Haven't tested with with `statusReciever: false`)
On Android, it only triggers it when using `statusReciever: true`.