randdusing/cordova-plugin-bluetoothle #575: how can use respond () ?

sigMonkey, 2020-03-05:
i try use my phone as Peripheral 
after a request read/ write i call a respond() but i have this error {"error":"connect","message":"No device address"} and after {"address":"4F:0C:D4:53:56:AE","name":null,"status":"disconnected"}

randdusing, 2020-03-07:
Can you paste some JavaScript code?

a-sac, 2020-04-20:
The same happens to me with respond or notify. I subscribe to `this.bluetoothle.initializePeripheral` and then I do:

```
if(ble.status === "subscribed") {
   this.bluetoothle.notify({
        "service":"1234",
        "characteristic":"ABCD",
        "value":"U3Vic2NyaWJlIEhlbGxvIFdvcmxk" //Subscribe Hello World
    ).then(r => console.log(JSON.stringify(r)));
}
```

It returns:

`zone-evergreen.js:651 Unhandled Promise rejection: {error: "connect", message: "No device address"}error: "connect"message: "No device address"__proto__: Object ; Zone: <root> ; Task: Promise.then ; Value: {error: "connect", message: "No device address"}`

randdusing, 2020-04-26:
Is this on Android? The device address needs to be sent as a param. I'll update the readme shortly

aparraga, 2020-04-26:
You mean in the respond? Same issue here: when call respond from an android I get the same error "No device address".
But I can't see an address param in the respond call. What can I do to solve it? Otherwise the write sequence is not processed, it result in write errors

a-sac, 2020-04-26:
Meanwhile, I read your examples, and I sent the device address as a param. It works perfectly!

marcui13, 2023-02-22:
Hi folks!

I'm trying to response to a 'writeRequested' status with respond() method.

My idea is that when you receive a message on either device, you can send a 'received' message to the device that sent that message.

I can see that the respond() method in the 'writeRequested' state has a success status, but i don´t know how to display the message 'Recived message in...' on the device sending the message.

The code of the my writeRequest function:

`writeRequest(devAddress:string, messageS:string){
    this.bluetoothle.write({
      address: devAddress,
      service: "1234",
      characteristic:"D6D95414-516C-4DCF-B5CF-3BB1A560AF0F",
      value: btoa(messageS),
      type: ""
    })
    .then((resp)=>{
      console.log('RESPONSE from writeRequest', resp);
    })
    .catch((err)=>{
      console.log('ERROR IN writeRequest:', err);
    })
  }`

The code of the writeRequested callback on peripheral device:

`if (status.status == "writeRequested") {
      if (
        status.service == "1234" &&
        status.characteristic == "D6D95414-516C-4DCF-B5CF-3BB1A560AF0F"
      ) {
        this.bluetoothle
          .respond({
            address: status.address,
            requestId: status.requestId,
            value: btoa('Recived message in tablet!') // encode base64
          })
          .then((data) => {
            console.log('RESPOND to writeRequested:', data);
            this.recivedMessages.push(atob(status.value)); // decode base64
            this.cdr.detectChanges();
            console.log(atob(status.value));
          })
          .catch((err) => {
            console.log('ERROR on RESPOND to writeRequested:', err);
          });
      }
    }`

Any idea please?

Regards, marcui :)