randdusing/cordova-plugin-bluetoothle #577: bug: Nullpointer Exception Crash Report

HansKrywaa, 2020-03-09:
Hey there,

we just published our App and i got a reported Crash now, which seems to be related to this Plugin. Maybe you can have a look into this? Stacktrace:

```
java.lang.NullPointerException: Attempt to invoke virtual method 'void android.bluetooth.le.BluetoothLeScanner.stopScan(android.bluetooth.le.ScanCallback)' on a null object reference

com.randdusing.bluetoothle.BluetoothLePlugin$1.onReceive BluetoothLePlugin.java:2784
android.app.LoadedApk$ReceiverDispatcher$Args.lambda$getRunnable$0 LoadedApk.java:1397
android.app.-$$Lambda$LoadedApk$ReceiverDispatcher$Args$_BumDX2UKsnxLVrE6UJsJZkotuA.run
android.os.Handler.handleCallback Handler.java:873
android.os.Handler.dispatchMessage Handler.java:99
android.os.Looper.loop Looper.java:280
android.app.ActivityThread.main ActivityThread.java:6706
java.lang.reflect.Method.invoke Method.java
com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run RuntimeInit.java:493
com.android.internal.os.ZygoteInit.main ZygoteInit.java:858
```

Happens to an User with **Xperia XA2** with Android 9.

randdusing, 2020-03-09:
Are you using the latest version? This should only be called when Bluetooth errors out (possibly turned off) and was already scanning.

randdusing, 2020-03-09:
This was recently merged: https://github.com/randdusing/cordova-plugin-bluetoothle/pull/456
Perhaps a null check just needs to be added. Some Android versions / phones may return null when BT is errored out.

HansKrywaa, 2020-03-09:
@randdusing In the App the Error appeared, Version `4.5.7` is used. I saw `4.5.8` came out yesterday, but i think the mentioned PR was already merged to `4.5.7`?

randdusing, 2020-03-11:
Do you have anyway to recreate this on your own, or do you have to involve a user?

HansKrywaa, 2020-03-11:
No i can not reproduce the Error 😕 I just see the Crash in the Reports...

But would it be just fixable to add `&& bluetoothAdapter.getBluetoothLeScanner() != null` to the If Statement on: ?
https://github.com/randdusing/cordova-plugin-bluetoothle/blob/8760eff8c84bc16924a42c1d91637a62dcf8c5c3/src/android/BluetoothLePlugin.java#L2783

randdusing, 2020-03-12:
Yep, easy to "fix". I just wanted to try to better understand why it's happening. I'll try to raise a fix soon.

HansKrywaa, 2020-03-12:
Ok understand. The Crash only appeared once so seems very rare 😕 Thank you!

randdusing, 2020-03-12:
Yah, it should be pretty rare. It would only occur if the user had started a scan and then Bluetooth turned off for whatever reason.

HansKrywaa, 2020-03-13:
Appeared again yesterday for a User with Xperia Z5 Compact and Android 7.1.1

randdusing, 2020-03-14:
Pushed a fix for this in 4.5.9. Let me know if it works

HansKrywaa, 2020-03-14:
Thank you 💪🏼 I will install it and use it in the next Release. I'm not sure when this will be published, as we are a big company and i can not decide that. So i will close this for now and open again if happens in the new Release again. Thanks for your investigation!

closed 2020-03-14 by HansKrywaa