randdusing/cordova-plugin-bluetoothle #578: bluetoothle.notify ERROR: "No device address"

ragcsalo, 2020-03-10:
(Android)
When I try to update a characteristic's value by using **bluetoothle.notify** I get this ERROR:
`"error":"connect", "message":"No device address"`

The documentation says: 
> Currently all subscribed devices will receive update. Device specific updates will be added in the future.

So far I could successfully set up the service, start advertising, and I could connect and subscribe from other devices. I just can't update the characteristic's value and notify subscribed devices. What am I missing?

randdusing, 2020-03-10:
Can you post some example code? Seems like you're not specifying the address

ragcsalo, 2020-03-11:
Thanks, I've figured it out in the meantime. :-) 

The documentation is wrong, it doesn't mention the "address" option among the parameters... also claims that every subscribed device will be notified, but in fact we have to do all the dirty work... :-(

I won't close this issue until they update the documentation.

randdusing, 2020-03-11:
Ahh looks like Android requires an address, but iOS doesn't. I'll update the documentation

randdusing, 2020-03-14:
Updated the docs

closed 2020-03-14 by randdusing