randdusing/cordova-plugin-bluetoothle #579: ACCESS_FINE_LOCATION is required for Android 10

alexhisen, 2020-03-19:
On Android, if you target SDK 29 (Android 10), then in hasPermission/requestPermission() you must use ACCESS_FINE_LOCATION instead of ACCESS_COARSE_LOCATION. The manifest should also request both of them.

randdusing, 2020-04-10:
It looks like there are also some permission changes required for scanning in the background. I don't have any Android devices to test on anymore, so hopefully someone else can raise a PR.

https://developer.android.com/about/versions/10/privacy/changes#app-access-device-location

lazydan, 2020-05-04:
The scan() function would work fine on Android 10 after i add the ACCESS_FINE_LOCATION permission to plugin.xml file.

NunoSav, 2020-06-16:
I've tried to change COARSE to FINE in the source code and it worked and then i tried changing plugin.xml seperately and it did not work.

PR #614 (2020-06-16): fix(Android 10): Use ACCESS_FINE_LOCATION instead of COARSE
This will fix the issue related to scanning devices in Android 10. I've tested with Android 6,7,9 and 10
fixes #579
closed 2020-06-16 by randdusing