randdusing/cordova-plugin-bluetoothle #580: manufacturerSpecificData on IOS ?

cmorillo, 2020-03-25:
Hi there,

Thanks for the plugin, is great, any way to pass on the manufacturerSpecificData on IOS?
It doesn't work for me and return the Service name.

Cheers,

Carlos

cmorillo, 2020-03-25:
advertisement = advertisement hash with the keys specified here (iOS)

ndrake, 2020-03-25:
This isn't possible as Apple's Core Bluetooth API doesn't allow specifying manufacturer specific data. 

https://developer.apple.com/documentation/corebluetooth/cbperipheralmanager/1393252-startadvertising

`An optional dictionary containing the data you want to advertise. The peripheral manager only supports two keys: CBAdvertisementDataLocalNameKey and CBAdvertisementDataServiceUUIDsKey.`

cmorillo, 2020-03-25:
Hi, in my case, I will be using the name property... that should do it.

closed 2020-03-25 by cmorillo