randdusing/cordova-plugin-bluetoothle #584: can't find BLE devices using scan.

sbom00, 2020-04-01:
I'm using an android (Moto g5+, Android 8.10), and Quasar/Cordova.

I Have already set up location services, permissions, and enabled Bluetooth. I had the program running and finding devices but did some modifications and lost functionality. When I did the rollback, it still didn't work. I have already uninstalled the app from my mobile, and have already restarted my computer, Android Studio and Visual Studio Code. I am at a loss in what to do now. 
Here is the code `connect: function(address) {

    log('Connecting to device: ' + address + "...", "status");

    if (cordova.platformId === "windows") {

        getDeviceServices(address);

    }
    else {

        stopScan();

        new Promise(function (resolve, reject) {

            bluetoothle.connect(resolve, reject, { address: address });

        }).then(connectSuccess, handleError);

    }
},
    addDevice: function(name, address) {
 
    var button = document.createElement("button");
    button.style.width = "100%";
    button.style.padding = "10px";
    button.style.fontSize = "16px";
    button.textContent = name + ": " + address;
 
    button.addEventListener("click", function () {
 
        document.getElementById("services").innerHTML = "";
        this.connect(address);
    });
 
    document.getElementById("devices").appendChild(button);
},
     initializeSuccess: function(result) {
 
    if (result.status === "enabled") {
 
        this.log("Bluetooth is enabled.");
        this.log(result);
        this.startScan();
    }
 
    else {
 
        document.getElementById("start-scan").disabled = true;
 
        this.log("Bluetooth is not enabled:", "status");
        this.log(result, "status");
    }
},
 handleError: function(error) {
 
    var msg;
 
    if (error.error && error.message) {
 
        var errorItems = [];
 
        if (error.service) {
 
            errorItems.push("service: " + (uuids[error.service] || error.service));
        }
 
        if (error.characteristic) {
 
            errorItems.push("characteristic: " + (uuids[error.characteristic] || error.characteristic));
        }
 
        msg = "Error on " + error.error + ": " + error.message + (errorItems.length && (" (" + errorItems.join(", ") + ")"));
    }
 
    else {
 
        msg = error;
    }
 
    this.log(msg, "error");
 
    if (error.error === "read" && error.service && error.characteristic) {
 
        reportValue(error.service, error.characteristic, "Error: " + error.message);
    }
},
 log: function(msg, level) {
 
    level = level || "log";
 
    if (typeof msg === "object") {
 
        msg = JSON.stringify(msg, null, "  ");
    }
 
    console.log(msg);
 
    if (level === "status" || level === "error") {
 
        var msgDiv = document.createElement("div");
        msgDiv.textContent = msg;
 
        if (level === "error") {
 
            msgDiv.style.color = "red";
        }
 
        msgDiv.style.padding = "5px 0";
        msgDiv.style.borderBottom = "rgb(192,192,192) solid 1px";
        document.getElementById("output").appendChild(msgDiv);
    }
},
retrieveConnectedSuccess: function(result) {

    this.log("retrieveConnectedSuccess()");
    this.log(result);

    result.forEach(function (device) {

        this.addDevice(device.name, device.address);

    });
},
 startScanSuccess: function(result) {

    this.log("startScanSuccess(" + result.status + ")");

    if (result.status === "scanStarted") {

        this.log("Scanning for devices (will continue to scan until you select a device)...", "status");
    }
    else if (result.status === "scanResult") {
        if (!this.foundDevices.some(function (device) {

            return device.address === result.address;

        })) {

            this.log('FOUND DEVICE:');
            this.log(result);
            this.foundDevices.push(result);
            this.addDevice(result.name, result.address);
        }
    }
},
requestPermissionSuccess: function (result){
  if (result==true){
    this.log("successfully requested permission?")
    bluetoothle.isLocationEnabled(this.isLocationEnabledSuccess, this.handleError);
  }else{
    this.log("request permission failed");
  }
  
},
hasPermissionResult: function(result){
  if (result == false){
    this.log("app does not have permission")
    bluetoothle.requestPermission(this.requestPermissionSuccess, this.handleError);
  }
  else{
    this.log("app has permition")
    bluetoothle.isLocationEnabled(this.isLocationEnabledSuccess, this.handleError);
    //
  }
},
isLocationEnabledSuccess: function(result){
  if (result==false){
    this.log("locations is not enabled")
    alert("please turn on location services");
    bluetoothle.requestLocation(this.log("user has been requested to enable location"), this.handleError);
    this.initializee();
  }
  else{
    this.log("location is enabled aleady");
    this.initializee();
  }
},
startScan: function() {
 
    this.log("Starting scan for devices...", "status");
 
    this.foundDevices = [];
 
    document.getElementById("devices").innerHTML = "";
    document.getElementById("services").innerHTML = "";
    document.getElementById("output").innerHTML = "";
    bluetoothle.startScan(this.startScanSuccess, this.handleError, { services: [] });
    
        
        
    
},
initializee: function(){
        
  new Promise(function (resolve) {
        bluetoothle.initialize(resolve, { request: true, statusReceiver: false });
 
    }).then(this.initializeSuccess, this.handleError);
},
    triggbtn: function () {
      this.log("button was pressed");
      this.t1 = performance.now();
      bluetoothle.stopScan(this.log('restarted scan'));
      bluetoothle.hasPermission(this.hasPermissionResult);     
    },
    
  },`

This is the log after I press the button

[image]

randdusing, 2020-04-04:
What type of device are you connecting to? Have you tried restarting that? And restarting the phone?

guspan, 2020-04-28:
Same problem

guspan, 2020-04-28:
Check this #579