randdusing/cordova-plugin-bluetoothle #585: advertise manufacturerData from iOS device

sapotrotas, 2020-04-02:
Hi @randdusing,
Gj with the plugin.

I am able to write **manufacturerData** in an encoded advertisement from an Androind device and then get and decode it just by using startScan(), but I'm not understanding how can I do the same from an iOS device since AdvertisingParamsIOS interface is only exposing "services" and "name" attributes. 
Is this possible, or do I have to create a service and a characteristic and then scan, discover and read?

Any help is appreciated,
Thanks!

randdusing, 2020-04-04:
iOS only supports name and services. https://developer.apple.com/documentation/corebluetooth/cbperipheralmanager/1393252-startadvertising

sapotrotas, 2020-04-04:
Thank you!

I'm closing the issue.

closed 2020-04-04 by sapotrotas