randdusing/cordova-plugin-bluetoothle #588: reconnect signature is wrong

chegewara, 2020-04-08:
This repository seems to be no longer maintained, but as a remainder and maybe if someone else will search for it. Reconnect function seems to have wrong signature. I found this is correct one:
```
reconnect(params: {
address: string;
}): Observable<
DeviceInfo
>;
```

randdusing, 2020-04-10:
I still "maintain" it, but not as much as I used to. PRs are appreciated!

randdusing, 2020-04-10:
Where is this an issue? This code base doesn't mention any observables.

chegewara, 2020-04-10:
Great, thanks for that plugin, i like it even if i am new to ionic.
I will try to create PR, its easy fix.

chegewara, 2020-04-10:
Its index.ts. Let me find it.

chegewara, 2020-04-10:
Path to file is:
`node_modules/@ionic-native/bluetooth-le/ngx/index.d.ts`

randdusing, 2020-04-10:
That's a wrapper for this plugin, so you'll need to raise the issue with them.

There code looks correct: https://github.com/ionic-team/ionic-native/blob/master/src/%40ionic-native/plugins/bluetooth-le/index.ts

chegewara, 2020-04-10:
This line:
https://github.com/ionic-team/ionic-native/blob/master/src/%40ionic-native/plugins/bluetooth-le/index.ts#L575

chegewara, 2020-04-10:
I did some debug and print logs and returned is straight DeviceInfo, not as status: DeviceInfo.

randdusing, 2020-04-10:
It's an issue with Ionic's plugin wrapper. I didn't notice the { status: DeviceInfo } the first time. You should raise a PR to fix it.

closed 2020-04-10 by randdusing

chegewara, 2020-04-10:
Great, thanks for explanation.