randdusing/cordova-plugin-bluetoothle #595: Seems it doesn't scan devices when phone's screen is off. Any solutions?

n1705771, 2020-04-26:
Hi Sir, 

I applied this plugin in my ionic 3 application. I enable the application working in background mode. The plungin's scan method works well when application is in both foreground and background. But when phone's screen turns off, it will stop scanning peripheral devices. Could you please tell how to let plugin works even when phone's screen turns off or is locked?

Many thanks

randdusing, 2020-04-26:
The behavior may have changed, but checkout this part of the docs: https://github.com/randdusing/cordova-plugin-bluetoothle#background-modes-ios

n1705771, 2020-04-29:
@randdusing Thanks

n1705771, 2020-04-29:
When my android phone's screen turns off, scanStart method only returns status: scanStarted and never return status: scanResult.

When phone's screen turns on, scanStart method can return status: scanResult whatever the app is in foreground or background. 

Below is the parameters setting, how to make scanStart method returns status: scanResult even through the phone's screen turns off?
 ```
var params = {
      "services": [],
      "allowDuplicates": false,
      "scanMode": this.bluetoothle.SCAN_MODE_LOW_LATENCY,
      "matchMode": this.bluetoothle.MATCH_MODE_AGRESSIVE,
      "matchNum": this.bluetoothle.MATCH_NUM_MAX_ADVERTISEMENT,
      "callbackType": this.bluetoothle.CALLBACK_TYPE_ALL_MATCHES,
    }
```

arnoudcommandeur, 2020-05-01:
I face same problem using Cordova with Android. Hope anyone has a solution.

randdusing, 2020-05-01:
Have you tried specifying a service UUID?

n1705771, 2020-05-02:
Yes. I put UUID service in params like:

```
var scanParams = {
      "services": [
        'faa1'
      ],
      "allowDuplicates": false,
      "scanMode": this.bluetoothle.SCAN_MODE_LOW_LATENCY,
      "matchMode": this.bluetoothle.MATCH_MODE_AGRESSIVE,
      "matchNum": this.bluetoothle.MATCH_NUM_MAX_ADVERTISEMENT,
      "callbackType": this.bluetoothle.CALLBACK_TYPE_ALL_MATCHES,
    }
```

n1705771, 2020-05-02:
And I find the same behavior happen in advertising method too. When phone's screen turns off, application stop advertising too.

arnoudcommandeur, 2020-05-02:
Maybe we should create a foreground service on Android?

n1705771, 2020-05-03:
I increase the scan time, scan method works well now. I also increase the advertising time, but when screen turns off, advertising works not very well as scan.

Both scan and advertising run in one application. The timer I set for scan is scan for 15 sec and stop for 5 sec; the timer for advertising is advertising for 10 sec and stop for 5 sec. 

Do you think this is a proper timeout setting?
Thanks

mattkhaw, 2020-05-21:
@n1705771 I think the issue here is not really related to the plugin but the limitation of the browser itself. Since the screen is off (most likely will be the same as background state), everything is suspended. But once you bring the app to the foreground, everything works as intended. Is that the behaviour on your current app? I'm also thinking on how to solve this issue.

randdusing, 2020-05-21:
This should be fixed thanks to @mattkhaw's PR: https://github.com/randdusing/cordova-plugin-bluetoothle/pull/607

mattkhaw, 2020-05-21:
@randdusing Oh really? That fixes this? Maybe I should do more testing on this.

randdusing, 2020-05-21:
Oops, I posted this on the wrong issue. This fixes #302

1416014977, 2025-01-14:
After android 8.1, if the screen is turned off, the scan requires a ScanFilter to perform the scan, but the cordova plugin-Bluetoothle plugin only creates a ScanFilter when a serviceuuid is given. So there are two ways to solve this problem, either by modifying BluetoothLePlugin.java to give an empty ScanFilter or by specifying one or more ServiceuuIds when using statScan()