randdusing/cordova-plugin-bluetoothle #598: iOS crashes on notify

a-sac, 2020-04-27:
Hey there,

I tried to connect two android devices using subscribe and notify. No problems so far.

When I try to connect an Android and an iPhone, the app crashes on notify. Everything is working except notify. And I don't even receive an error, the app simply crashes.

The code:

```
subscribed(ble) {
    this.setMessage(ble.address + " subscribed");

    var params = {
      "service":"1234",
      "characteristic":"ABCD",
      "value": "V3JpdGUgSGVsbG8gV29ybGQ"
    };

    params["address"] = ble.address;

    this.bluetoothle.notify(params).then(
      response => this.setMessage(JSON.stringify(response)),
      error => this.setMessage(JSON.stringify(error))
    );
  }
```

randdusing, 2020-04-28:
Any logs from the crash?

a-sac, 2020-04-28:
@randdusing 
`MyApp[76071:804074] *** Terminating app due to uncaught exception 'NSInternalInconsistencyException', reason: 'Invalid parameter not satisfying: value != nil'`

a-sac, 2020-04-28:
Xcode redirects me to **line 460** of 

> BluettothPlugin.m 

` BOOL result = [peripheralManager updateValue:value forCharacteristic:checkCharacteristic onSubscribedCentrals:nil]; //TODO need to store CBMutableCharacteristic `

Sorry if I'm doing something stupid, I'm new to all Ionic/ Mobile ecosystem :/

randdusing, 2020-05-01:
This may be a base64 encoding issue. There should be an equals sign at the end

a-sac, 2020-05-08:
@randdusing it worked, thanks!

randdusing, 2020-05-09:
The readme values look correct, not all of them end in an equals, but that one did :)

closed 2020-05-12 by a-sac