randdusing/cordova-plugin-bluetoothle #599: Create Central and Peripheral in one

arnoudcommandeur, 2020-05-01:
Hello, i want to create an app that is both central and peripheral at the same time. Is this possible?
Cheers, 
Arnoud

randdusing, 2020-05-01:
I remember trying this out once but can't remember the results. It should be easy to test out :)

arnoudcommandeur, 2020-05-02:
Looks like it's working @randdusing  Unfortunately I see a address 02:00:00:00:00:00 when I execute bluetoothle.getAdapterInfo(log);

Is this correct behaviour? I can't find the peripheral when I scan on a different device.

index.js:188 {
  "address": "02:00:00:00:00:00",
  "name": "Galaxy S10",
  "isInitialized": true,
  "isEnabled": true,
  "isScanning": false,
  "isDiscoverable": false
}

Thanks for your answer in advance...

Arnoud

arnoudcommandeur, 2020-05-02:
Hi @randdusing I just found out that this is expected behaviour (the address). Still not able to see the device...

arnoudcommandeur, 2020-05-02:
Extra info @randdusing , I created the peripheral as following:

        var params = {
        service: "1234",
        characteristics: [
            {
            uuid: "ABCD",
            permissions: {
                read: true,
                write: true,
                //readEncryptionRequired: true,
                //writeEncryptionRequired: true,
            },
            properties : {
                read: true,
                writeWithoutResponse: true,
                write: true,
                notify: true,
                indicate: true,
                //authenticatedSignedWrites: true,
                //notifyEncryptionRequired: true,
                //indicateEncryptionRequired: true,
            }
            }
        ]
        };
        bluetoothle.addService(successAS, errorAS,params);

function successAS(result) {
    log(result);
    bluetoothle.startAdvertising(log, log, {"service":"1234","name":"Hello World"});   
}

function errorAS(result) {
    log(result);
}

arnoudcommandeur, 2020-05-02:
The Antivirus software blocked the bluetooth on the peripheral, now I can see my device when I am scanning....

closed 2020-05-02 by arnoudcommandeur

marcui13, 2023-02-03:
Hi, how are you??
Were you able to get your Central/Peripheral app up and running? 
I'm trying to develop something similar and I can't make it work for me. 
Could you show me how you did it please??