randdusing/cordova-plugin-bluetoothle #600: Set name of peripheral

arnoudcommandeur, 2020-05-02:
Hi all, now can I set the name of the peripheral for Android device running the app? I only see the name of my phone when i am scanning. 

Thanks

randdusing, 2020-05-03:
You should be able to pass a name parameter to the startAdvertising() method.

arnoudcommandeur, 2020-05-04:
Hi @randdusing I already, but no difference:

    bluetoothle.startAdvertising(log, log, {"service":"1234","name":"Hello World","timeout":180000});   

Do you have another suggestion?

sbhm2, 2020-05-06:
I am trying to run my app as central and peripheral at the same time. The central and peripheral roles are established properly. But the problem I am facing is that the peripheral name is gets set to some random 8 characters e.g., "b5803c96", "89ee180f" on different devices. @randdusing do you have any idea what is exactly going on. I have used the exact codes from github.

arnoudcommandeur, 2020-05-07:
@sbhm2 which phone do you use?

sbhm2, 2020-05-07:
I have used Redmi Note 5(Android 8), where I got the name(on .getAdaptorInfo() and on scan) 89ee180f and Redmi Note 6 Pro(Android 9), where I got the name(on .getAdaptorInfo() and on scan) b5803c96. Also in both cases, the default device Bluetooth name was different.

sbhm2, 2020-05-07:
> @sbhm2 which phone do you use?

I have used Redmi Note 5(Android 8), where I got the name(on .getAdaptorInfo() and on scan) 89ee180f and Redmi Note 6 Pro(Android 9), where I got the name(on .getAdaptorInfo() and on scan) b5803c96. Also in both cases, the default device Bluetooth name was different.