randdusing/cordova-plugin-bluetoothle #602: Recieve TxPowerLevel from Peripheral on Android

arnoudcommandeur, 2020-05-04:
Hi @randdusing when I inspect the ScanResult object from an Peripheral, I want to get the TxPower. Is this only sent by iOS? And is this the same as txPowerLevel?

{
  "status": "scanResult",
  "advertisement": "awArG05L", //Android
  "advertisement": { //iOS
    "serviceUuids": [
      "180D"
    ],
    "manufacturerData": "awAvFFZY",
    "txPowerLevel": 0,
    "overflowServiceUuids": [
    ],
    "isConnectable": true,
    "solicitedServiceUuids": [
    ],
    "serviceData": {
    },
    "localName": "Polar H7 3B321015"
  },
  "rssi": -58,
  "name": "Polar H7 3B321015",
  "address": "ECC037FD-72AE-AFC5-9213-CA785B3B5C63"
} 

Hope to hear from toy, thanks again!

randdusing, 2020-05-07:
Android only returns back the advertisement encoded string of byte. You'll have to parse that, but it most likely doesn't return what you need.

arnoudcommandeur, 2020-05-07:
Ok @randdusing  thanks. I will investigate it further.