randdusing/cordova-plugin-bluetoothle #603: Reading encrypted characteristic fails on iOS even after pairing

hschultz1960, 2020-05-06:
### Issue

On iOS, connecting to a _previously paired device_ fails. The `connect()` function never returns in any of the callbacks, it hangs forever.

### Details

My issue concerns being able to read and write characteristics, where the peripheral requests _encryption_ to be enabled on the connection. The BT spec explains how this is done at the BLE protocol level, commonly referred to as "just works" security. This implicitly requires pairing to take place between the central and the peripheral, and is something that the OS automatically handles for us within the BLE stack.

The following sequence seems to be the accepted approach to perform implicit pairing at the BLE level. With the peripheral put into _pairing mode_:

```
scan() for peripheral
connect() to peripheral
read() characteristic requiring encrypted transport
disconnect()
```

At the point of the `read()` call, the OS automatically opens a native dialog, to allow the user to accept or reject the device to be paired. Afterwards, _without_ the peripheral being in pairing mode, the exact same sequence can be used again, _without_ the OS requiring the user to pair. 

And, in fact, this works fine all the way on Android.

On iOS however, later attempts to access the peripheral fails. Initially, the pairing dialog is shown and the pairing succeeds. The peripheral also shows up in the list of paired devices under iOS Settings. `scan()` also shows the device. But `connect()` attempts _hang infinitely_, whether or not it is done in the same app session, or when the app later is reopened.

It is worth noting, that _without_ the peripheral being paired, `connect()` always succeeds (but of course any subsequent reading or writing fails due to lack of encryption).

Something is clearly wrong here. There must be something in the way the device, once paired, is being connected to, that makes the iOS BLE stack hang.

### My ask

Anybody else seen this behavior?
Do we know for sure if this plugin supports the use of paired devices on iOS?
Is there anything in the above sequence that should be done differently for iOS, to correctly pick up a previously paired device?
Is there a way that I can enable detailed debugging in the BLE stack, perhaps from within Xcode, that can help diagnose the issue? 

Thanks for any insight anybody can provide.

davflo-4Soft, 2020-08-05:
I am experiencing the same behavior. First connection works fine with pairing dialog. For any further connection requests (i.e. app has been closed and then reopened) no callback is received.
@hschultz1960 Did you find out why?

hschultz1960, 2020-08-05:
No, I am still having this issue. My customer has accepted that we team up with an experienced iOS developer, to help look into the plugin from the OS side. We are awaiting the summer holiday period to end before resuming work, probably in a couple of weeks (mid August).

DalterioRaffaele, 2022-07-14:
I have this problem, I use ios as Central and Android as Peripheral. I do these steps on ios:                                                    
scan () for peripheral
connect () to peripheral
read () characteristic
disconnect ().

After disconnecting if I do a scan again I can no longer connect. Anyone have a solution? Thanks