randdusing/cordova-plugin-bluetoothle #604: How to increase the size of manufacturerSpecificData in ble peripheral advertisement?

sbhm2, 2020-05-07:
I am trying to use peripheral and central on the same device at the same time and I am pretty much successful in doing that. On the peripheral side I am adding a service with id "1234" and then added manufacturerSpecificData:[encodedString]. On the central side, I am can successfully extract the manufacturerSpecificData from the advertisement packet.
 Thank you @randdusing for the plugin and extensive support I got through GitHub.
But one point where I got stuck and put head for almost a day, is I am unable to add more than 8 characters long string in manufacturerSpecificData. What I have read about ble packet is I can add around 28-29bytes of data. Then why I am getting **{error: "startAdvertising", message: "Too large data"}** on advertising a packet containing manufacturerSpecificData more than 8 chars long string? How can I increase the length of this string?
Here is my startAdvertisement code.
[image]

randdusing, 2020-05-07:
Which type of device is this? I think their are some OS specific restrictions on size of manufacturer data

hschultz1960, 2020-05-07:
The advertising packet can only contain up to 31 bytes. This is a hard limit per the BT spec. And, you can't have them all to yourself ;-) 

This [article](https://www.silabs.com/community/wireless/bluetooth/knowledge-base.entry.html/2017/02/10/bluetooth_advertisin-hGsf)  explains the advertising packet format in much detail.

sbhm2, 2020-05-07:
> Which type of device is this? I think their are some OS specific restrictions on size of manufacturer data

I am testing it on android mobile.

sbhm2, 2020-05-07:
> Which type of device is this? I think their are some OS specific restrictions on size of manufacturer data

I am using simple android devices for testing.

sbhm2, 2020-05-07:
> The advertising packet payload can only contain up to 31 bytes. This is a hard limit per the BT spec. And, you can't have them all to yourself ;-)
> 
> This [article](https://www.silabs.com/community/wireless/bluetooth/knowledge-base.entry.html/2017/02/10/bluetooth_advertisin-hGsf) explains the advertising packet format in much detail.

Thank you @hschultz1960 sir, for the article. But clearly I have the liberty to send 28 bytes. And my problem is I am only able to send 7 bytes in manufacturerSpecificData(). I am unable to understand from where the remaining bytes are getting filled. Also this is the advertisement data I am getting on scan. There are lots of "A" at the end. what are those and is there any way to have control over this.
[image]

sbhm2, 2020-05-08:
I have managed to successfully do this. Thanks to [](http://www.drjukka.com/blog/wordpress/?p=154).
I set {
      service: "1234"
      includeDeviceName: false,
      includeTxPowerLevel: false,
      connectable: false,
      manufacturerSpecificData: bluetoothle.bytesToEncodedString(Buffer.from("MY 20BYTE HEX DATA", "hex"))
}
in advertising params. 
And on successful scanning, Buffer.from(bluetoothle.encodedStringToBytes(result.advertisement)).toString('hex') gives me raw data which contains my manufacturerSpecificData along with serviceID.

closed 2020-05-08 by sbhm2