randdusing/cordova-plugin-bluetoothle #605: Unable simply Scan from a Clean-started project with Ionic-Capacitor

user1337, 2020-05-10:
Hi there!

It there anyone who is capable to do a Ionic project with Capacitor 
that is capable of doing a Scan and getting "scanResult" ?
I also got a success from "scanStarted".

I started by following the steps for Capacitor installation  from :
   https://ionicframework.com/docs/native/bluetooth-le

What I also did was :

- Before do "startScan" I also ask for permissions before :
   requestPermission()
   requestLocation()
 
- Add bluetooth permissions to AndroidManifest.xml :
   <uses-permission android:name="android.permission.BLUETOOTH" />
   <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />

- Set the SdkVersion of "variables.gradle" :
    minSdkVersion = 23
    compileSdkVersion = 29
    targetSdkVersion = 29

Thanks,
Huy

user1337, 2020-05-10:
> Hi there!
> 
> It there anyone who is capable to do a Ionic project with Capacitor
> that is capable of doing a Scan and getting "scanResult" ?
> I also got a success from "scanStarted".
> 
> I started by following the steps for Capacitor installation from :
> https://ionicframework.com/docs/native/bluetooth-le
> 
> Otherwise, I also start a Ionic project with "Cordova" and the Scan working
> and I get scan results with some devices.
> 
> What I also did was :
> 
> * Before do "startScan" I also ask for permissions before :
>   requestPermission()
>   requestLocation()
> * Add bluetooth permissions to AndroidManifest.xml :
>   android.permission.BLUETOOTH
>   android.permission.BLUETOOTH_ADMIN
>   android.permission.ACCESS_FINE_LOCATION
> * Set the SdkVersion of "variables.gradle" :
>   minSdkVersion = 23
>   compileSdkVersion = 29
>   targetSdkVersion = 29
> 
> Thanks,
> Huy

BetterAutomations, 2020-07-07:
I tried the same and received this error in the console. Is that what you received?

`2020-07-07 05:55:15.400 4026-4040/io.ionic.starter W/Binder: Binder call failed.
    java.lang.SecurityException: Need ACCESS_COARSE_LOCATION or ACCESS_FINE_LOCATION permission to get scan results
        at android.os.Parcel.readException(Parcel.java:1693)
        at android.os.Parcel.readException(Parcel.java:1646)
        at android.bluetooth.IBluetoothGatt$Stub$Proxy.startScan(IBluetoothGatt.java:796)
        at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper.onClientRegistered(BluetoothLeScanner.java:423)
        at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:56)
        at android.os.Binder.execTransact(Binder.java:573)
`

randdusing, 2020-07-14:
Are you calling requestPermission() before hand?