randdusing/cordova-plugin-bluetoothle #606: Android: stopAdvertising is called automatically after startAdvertising

JorgeMorata, 2020-05-14:
Good Morning,

I am using the library to develop an application that allows using the mobile phone as a BLE and at the same time make scans of other BLEs around that have the same application.

To do this, I perform startAdvertising (which works on both iOS and Android) and then scans that also return the other mobile phones well.

The problem comes, since in Android, immediately after performing startAdvertising (and for example from an iOS detect the android phone), the Android phone is no longer visible.

The application performs a startScan every 30 seconds to find new BLE devices (while the adverstising is on)

Checking the android system logs I see that a call is made to stopAdvertising but it is not made from my code. It seems to be called from the system.

The traces I have found are the following in case they help in the diagnosis:

```
05-14 01:30:47.968  4458  4458 I chromium: [INFO:CONSOLE(1114)] "Bluetooth Peripheral Advertising started: {"mode":1,"timeout":1000,"txPowerLevel":3,"isConnectable":false,"status":"advertisingStarted"}"
05-14 01:30:48.048   805   805 I aptouch_daemon: check_afe_status, idle in. 100
05-14 01:30:48.224  4458  4541 D NetworkSecurityConfig: Using Network Security Config from resource network_security_config debugBuild: true
05-14 01:30:48.674  4458  4458 I BluetoothLeAdvertiser: stopAdvertising is called
05-14 01:30:48.702  1217  1806 I TrafficMonitor: update:rxPkts:29,txPkts:35,rxBytes:20049,txBytes:5124
05-14 01:30:48.702  1217  1806 I TrafficMonitor: no expired. level:2
05-14 01:30:48.702  1217  1806 I TrafficMonitor: gettimer:interval=2000
05-14 01:30:48.703  1217  1806 I TrafficMonitor: during_ms:2002,period_ms:2000
05-14 01:30:48.703  1217  1806 I TrafficMonitor: count:226,rx_sum:29,tx_sum:35,rxBytes:20049,during_ms:2002,rx_speed:10014.485514485516,tx_speed:2559.44055944056,rto:0.8285714285714286
05-14 01:30:48.734  4458  4541 D SERVER  : Handling local request: http://localhost/33-es2015.js
05-14 01:30:48.736  4458  4541 D SERVER  : Handling local request: http://localhost/75-es2015.js
05-14 01:30:48.737  4458  4541 D SERVER  : Handling local request: http://localhost/69-es2015.js
05-14 01:30:48.738  4458  4541 D SERVER  : Handling local request: http://localhost/29-es2015.js
05-14 01:30:48.738  4458  4542 D SERVER  : Handling local request: http://localhost/17-es2015.js
05-14 01:30:48.765  2139  3458 D ConnCureStateMonitor: handleMessage: msg[4]
05-14 01:30:48.767  2139  3458 D ConnCureStateMonitor: canPerformConnCure: wifi link normally
05-14 01:30:48.906 21705 24945 E bt_stack: [ERROR:btm_ble_multi_adv.cc(645)] Invalid or no active instance
```

Is it possible that if a startScan is performed while advertising is on, stopAdvertising is called on Android?

If more information is needed, do not hesitate to request it.

Thank you.

mattkhaw, 2020-05-21:
@JorgeMorata I'll throw in my 2 cents here. I'm also in the midst in developing an app which uses both central and peripheral functions on the same device and both functions are running in tandem as well. I'm doing scanning and advertising at the same time. It seems to be working fine. I'm using Android 6 btw. Maybe it is something else that is causing issues?

JorgeMorata, 2020-06-02:
Hi! 

I don't know... In iOS works perfectly, and scanning from an Android Device, find the device correctly, but when the device i'm trying to scan is Android, only in the first milisiconds the service/device is available to others devices.

On the occasions the device is founded, i'm trying to connect to it automatically and it's already stopped/dissapeared (so is stopped in milliseconds).

Thank for your response!

mattkhaw, 2020-06-02:
Weird indeed. I've only tested between iOS and Android. I haven't test Android with Android or iOS with iOS due to the limited amount of devices I have. Maybe I'll try it out and see.

WWWMARIO, 2020-07-15:
Your advertising timeout is set to 1000 ms.

Try setting your AdvertisingParams to a higher timeout.

export interface AdvertisingParamsAndroid {
    service: string;
    mode?: AdvertiseMode;
    connectable?: boolean;
    **_timeout?: number;_**
    txPowerLevel?: TxPowerLevel;
    manufacturerId?: number;
    manufacturerSpecificData?: any;
    includeDeviceName: boolean;
    includeTxPowerLevel: boolean;
}

JorgeMorata, 2020-07-19:
> Your advertising timeout is set to 1000 ms.
> 05-14 01:30:47.968 4458 4458 I chromium: [INFO:CONSOLE(1114)] "Bluetooth Peripheral Advertising started: {"mode":1,"**_timeout":1000_**,"txPowerLevel":3,"isConnectable":false,"status":"advertisingStarted"}"
> 
> Try setting your AdvertisingParams to a higher timeout.
> 
> export interface AdvertisingParamsAndroid {
> service: string;
> mode?: AdvertiseMode;
> connectable?: boolean;
> **_timeout?: number;_**
> txPowerLevel?: TxPowerLevel;
> manufacturerId?: number;
> manufacturerSpecificData?: any;
> includeDeviceName: boolean;
> includeTxPowerLevel: boolean;
> }

F****!!! I didn't set the timeout and i thought it was "0"... Thank you for your response! Probably it fix the problem (Because of the waiting, I finally did the app Android Native and it works perfectly so I discontinued the Ionic Android App).

Thank to all for the help!

closed 2020-07-19 by JorgeMorata

asif166037, 2025-10-17:
Thank you so much @jmoratat !!