randdusing/cordova-plugin-bluetoothle #615: Many NullPointer AppCrashes

HansKrywaa, 2020-06-19:
We received many many Crashes caused of this Plugin by a NullPointer Exception:

**java.lang.NullPointerException: Attempt to invoke virtual method 'void android.bluetooth.BluetoothGattServer.close()' on a null object reference**

```
com.randdusing.bluetoothle.BluetoothLePlugin$1.onReceive BluetoothLePlugin.java:2798
android.app.LoadedApk$ReceiverDispatcher$Args.lambda$getRunnable$0$LoadedApk$ReceiverDispatcher$Args LoadedApk.java:1646
android.app.-$$Lambda$LoadedApk$ReceiverDispatcher$Args$_BumDX2UKsnxLVrE6UJsJZkotuA.run
android.os.Handler.handleCallback Handler.java:883
android.os.Handler.dispatchMessage Handler.java:100
android.os.Looper.loop Looper.java:237
android.app.ActivityThread.main ActivityThread.java:8019
java.lang.reflect.Method.invoke Method.java
com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run RuntimeInit.java:493
com.android.internal.os.ZygoteInit.main ZygoteInit.java:1100
```

[image]

randdusing, 2020-06-20:
Looks to be related to this change: https://github.com/randdusing/cordova-plugin-bluetoothle/commit/4382eb0a5a2f2c3e5345887f382187d33ea7f722
Looks like TURNING_OFF can be triggered when not expected. A simple null check should suffice.

HansKrywaa, 2020-06-22:
Is it possible that you can release a quick fix Version for this?

randdusing, 2020-06-23:
PRs are always welcome, but I may be able to raise one myself this week

HansKrywaa, 2020-06-24:
Just create the PR. That was not possible for my Java knowledge😃

randdusing, 2020-06-25:
I've released 4.15.14, so this should be fixed now.

HansKrywaa, 2020-06-25:
I was able to force the Crash by toggling Airplane Mode. This doesn't crashed now to think it works. Thanks 😊

closed 2020-06-25 by HansKrywaa