randdusing/cordova-plugin-bluetoothle #621: subscribedResult Never received

xVishera, 2020-07-01:
Hello,

I'm trying to use that plugin with Ionic 5.

I wrote some code that successfully connect to my device (weight measuring and oximeter).

I successfully receive the status "connected"

After that, I tried to subscribe to the callbacks that I want, 

I successfully receive the status "subscribed"

After that, I put some weight on the measurer or my finger in the oximeter, it read all the information and signals that is sending the information ( bluetooth icon ), but I never receive another callback in my subscription with the subscribedResult

Currently, I'm doing the following flow

> Initialize
>  Connect
> Discover
> Subscribe

If I try to call a "bond" request, is always returns "bonding" and then "unbounded".

Can anyone tell me if I'm forgetting something?

Thanks

```
         import { Component, OnInit } from '@angular/core';
         import { BluetoothLE, CurrConnectionStatus } from "@ionic-native/bluetooth-le/ngx";

        @Component({
        selector: 'app-test-connect-ble',
        templateUrl: './test-connect-ble.page.html',
        styleUrls: ['./test-connect-ble.page.scss'],
        })
        export class TestConnectBLEPage implements OnInit {

        deviceId = "9C:1D:XX:XX:XX:XX"; //mac

        constructor(private bluetoothLE: BluetoothLE) { }

        ngOnInit() {
        }

        Scan() {
            this.bluetoothLE.initialize({statusReceiver: true, request: true, restoreKey: "mytest"}).subscribe((data) => {
            console.log("Inicialize" + JSON.stringify(data));

            this.bluetoothLE.connect({ address: this.deviceId })
                .subscribe((dataConect) => {
                console.log("Conexão: " + JSON.stringify(dataConect));

                if (dataConect.status === "connected") {
                    this.readData();
                }
                });
            });
        }

        readData() {

            this.bluetoothLE.discover({ address: this.deviceId }).then((dataDis) => {
            console.log('Dados Discovery: ' + JSON.stringify(dataDis));
            console.log('-------------------------------------------');
            this.testeSub('7802', '8a23');
            this.testeSub('7802', '8a82');
            });
        }

        ReadAnything() {
            this.bluetoothLE.read({
            address: this.deviceId,
            service: '7802',
            characteristic: '8a82'
            }).then(response => {
            console.log('Read response: ' + JSON.stringify(response));
            var bytes = this.bluetoothLE.encodedStringToBytes(response.value);
            var result = this.bluetoothLE.bytesToString(bytes);
            var resultEncode = this.bluetoothLE.bytesToEncodedString(bytes);
            console.log(`Service: ${response.service},characteristic: ${response.characteristic},  Value64: ${response.value}, ValueString: ${result}, valueEncode: ${resultEncode}`);

            }).catch(err => {
            console.log('Read err: ' + JSON.stringify(err));
            });
        }

        testeSub(service, characteristic) {
            this.bluetoothLE.subscribe({
            address: this.deviceId,
            service: service,
            characteristic: characteristic
            }).subscribe((next) => {
            console.log('next: ' + JSON.stringify(next));
            }, (error) => {
            console.log('error: ' + JSON.stringify(error));
            }),
            () => {
                console.log("rodou o complete");
            }
        };

        subscribeToMethods() {
            this.bluetoothLE.subscribe({
            address: this.deviceId,
            service: 'BA11F08C-5F14-0B0D-1080-007CBE938D3C',
            characteristic: 'CD04'
            }).subscribe(dataSub => {

            console.log('Dados subscribe: ' + JSON.stringify(dataSub));

            // Se receber dados
            // if (dataSub.status == 'subscribedResult'){
            //  var bytes = this.bluetoothLE.encodedStringToBytes(dataSub.value);
            //  var result = this.bluetoothLE.bytesToString(bytes);
            //  var resultEncode = this.bluetoothLE.bytesToEncodedString(bytes);
            // console.log(`Service: ${dataSub.service},characteristic: ${dataSub.characteristic},  Value64: ${dataSub.value}, ValueString: ${result}, valueEncode: ${resultEncode}`);
            // }
            });
        }

        }
`

alexander090496, 2020-07-03:
When you connect or subscribe in this line "console.log ('Dice subscribe:' + JSON.stringify (dataSub));" gives you as a "subscribed" status? if not the delivery this state is for the characteristic this badly sent or to consult the characteristic well

As an example I have a device that gives me this when I subscribe

D / SystemWebChromeClient: http: //localhost/entities-belista-belista-module-es2015.js: Line 558: {"address": "EB: 5D: 75: 85: 32: 0D", "name": "T1S "," service ":" 6E400001-B5A3-F393-E0A9-E50E24DCCA9E "," feature ":" 6E400003-B5A3-F393-E0A9-E50E24DCCA9E "," descriptor ":" 2902 "," status ":" subscribed "}

after that the notification is left until the data can be received

xVishera, 2020-07-03:
> When you connect or subscribe in this line "console.log ('Dice subscribe:' + JSON.stringify (dataSub));" gives you as a "subscribed" status? if not the delivery this state is for the characteristic [...]

@alexander090496 Yes, I receive the callback with "subscribed" status, just like your example. After that, I never receive an callback of subscribeResult.

I'm doing some tests and I think the problem is related with paring the device. I think that I have to "write" some kind of authorization code or anything like that.

alexander090496, 2020-07-03:
ok, i have the same problem with a device, he has his native application and makes that connection or authentication, could you tell me how that authentication is done in the writer

xVishera, 2020-07-03:
> 
> 
> ok, i have the same problem with a device, he has his native application and makes that connection or authentication, could you tell me how that authentication is done in the writer

Thats the problem, I'm trying to get the technical documentation from the manufacter to know how to do the authentication. Atm I have an java and IOS SDK, but the source is obscured.

ABRT29, 2020-07-26:
Did you find a solution ? Thank you

sjennen, 2020-09-15:
I had the same problem. I was originally using a promise for the call to bluetoothle.subscribe, but when I call it directly it works and I get subscribedResults.

   bluetoothle.subscribe(subscribeSuccess, subscribeError, { address: myAddress, service: serviceUuid, characteristic: characteristicUuidTx});