randdusing/cordova-plugin-bluetoothle #623: Request Permission

willishf, 2020-07-03:
Using the API to do some testing and noticed that if I completely uninstall the app that was working prior and ask bluetoothle.isLocationEnabled(locationRequestPermission, handleError); will get a true returned so don't make a call to bluetoothle.requestPermission(locationRequestionPermissionSuccess, handleError); under the assumption everything is good. Go through and do the startScan and it reports it is scanning but get no events.

If I manually call bluetoothle.requestPermission(locationRequestionPermissionSuccess, handleError); via a separate button action will get prompted to give permission. The blue tooth init code will now work and I will get Bluetooth events. Quit the app and come back in and everything works. Ask again via the button to request permission do not get the pop up asking user for permission.

The bug seems to be triggered by an uninstall and then reinstall of the app.

Does appear can be solved by always asking for permission.

randdusing, 2020-07-03:
Why not query hasPermission first and then query requestPermission if needed? After that, see if location is enabled. The popup will only appear once per app install or else developers could spam a user.

willishf, 2020-07-03:
Somehow I mentally associated isLocationEnabled as the hasPermission for
requestPermission.

Thanks for the quick response.