randdusing/cordova-plugin-bluetoothle #626: statusReceiver explanation unclear. (What does that mean?)

Juarrow, 2020-07-07:
In the API description of the `initialize`-method, it says for parameter `statusReceiver`:

> Should change in Bluetooth status notifications be sent

What does that mean?
- (If enabled) Does this mean, that the paramter changes, when BT status notifications are received?
- (If enabled) Does ist mean, that the callback is called when the BT status changes?
- etc.

I'm sorry, but this "English" does not make any sense to me.

randdusing, 2020-07-14:
The naming is a bit bad. It should really be called "shouldReceiveStatusUpdates". Basically you call #initialize() once and you'll receive a promise notification every time the status changes (BT off, BT on, etc)

Related code:
https://github.com/randdusing/cordova-plugin-bluetoothle/blob/master/src/android/BluetoothLePlugin.java#L2762
https://github.com/randdusing/cordova-plugin-bluetoothle/blob/master/src/ios/BluetoothLePlugin.m#L2018

Juarrow, 2020-07-15:
Thank you for the clarification.

closed 2020-07-15 by Juarrow