randdusing/cordova-plugin-bluetoothle #629: Bluetooth background scanning not working when the app is background in cordova application

Abhinav4952, 2020-07-14:
I am initializing the blueeothle and peripheral using the foll code:-

```
bluetoothle.initialize(function initializeResult(result) {
        console.log(result);
      },  {
        request: true,
        statusReceiver: true,
        restoreKey: "bluetoothleplugin",
      });
     bluetoothle.initializePeripheral(
        function initializeResult(result) {
          console.log("initialize peripheral");
          console.log(result);
        },
        function error(result) {
          console.log("init peripheral error");
          console.log(result);
        },
        {
          request: true,
          restoreKey: "bluetoothleplugin",
        }
      );
          
```

and I started the scan using the foll options:-
```
bluetoothle.startScan(
        (result: BluetoothScanResult) => {
          console.log("result in searching");
          // console.log(result);
          if (result.name) {
            bluetoothresult.push(result);
            resulllt.push(result);
            bluetoothresultconst.push(result);
            resullltconst.push(result);
          }
        },
        function startScanErrorCallback(err) {
          console.log("printing err");
          console.log(err);
        },
        { serviceUuids: [],allowDuplicates:true }
      );

      setTimeout(() => {
        bluetoothle.stopScan(
          (result) => {
            console.log("stopping the scanning");
            console.log(bluetoothresult);
            if (bluetoothresult.length != 0) {
              console.log("got result length is more than zero");
                }
          },
          function stopScanErrorCallback(result) {
            console.log("came error in stopping the scanning ");
            console.log(result);
          }
        );
      }, 40000);
```
I am getting the scan results fine but when i moved the application to background using the following code:-
`cordova.plugins.backgroundMode.moveToBackground();`

i was getting empty results for the scan .Ive kept the interval for scan as 60000ms and 40000ms timeout for stopping
but immediately when the app is made to run in foreground scan results were fine.

**Note:-**I am cheking the applicaiton in a Android 9 device ive even added permission for foreground service in manifest.xml of the cordova and plugin.xml also

randdusing, 2020-07-21:
Do you need to call #initializePeripheral()? That may have odd behavior in background mode.

Also, try scanning by a particular service while in the background. I believe iOS requires it and possibly Android does now as well.

mattkhaw, 2020-07-30:
Unfortunately, this plugin doesn't support background callbacks. The issue is not the native code but more on Javascript side. This is by design unfortunately. I've suggested this before but I haven't got the time to try that out so, that issue is here by the way #608 

So, if you have time, you can try modifying the current plugin to accommodate this and post back here. For a better idea, I think can check out cordova-plugin-beacon (I think that's the name).

As for background scanning, iOS doesn't support custom UUIDs while, Android does everything. I think these limitations are documented in the plugin's README section.

randdusing, 2020-07-31:
What do you mean by background callbacks? This app can support a connection running in the background

mattkhaw, 2020-08-03:
@randdusing The plugin is working as intended. However, the callbacks will only kick in once the app is resumed. I'm proposing to change that by using this directive instead of the current one:

`[self.commandDelegate runInBackground:^{ // Logic }];`

This will allow the plugin to callback to the web view when it is in the background. I've looked into the cordova-plugin-beacon plugin's code and found that. That's how they are doing it. Of course, this only can happen when combined with WKWebView since it has the capability of running scripts in the background.

1416014977, 2025-01-14:
If you need to scan through the background, you need to keep scanning. If you run statScan() again in the background using settimeout or setinterval, you won't get the callback result this time