randdusing/cordova-plugin-bluetoothle #630: Bluetooth connection in background

ABRT29, 2020-07-21:
hello, my bluetooth button only has one button. when I click on this button it emits information. This is very simple.

To know if this one emitted information I am currently doing a loop every 10 seconds which tries to connect (use of connect ()) if it succeeds then the button emitted. Unfortunately it does not work in the background, after 10 minutes it stops working ...

How can I improve my verification process?

Thank you

ABRT29, 2020-07-26:
Any ideas, please ?

randdusing, 2020-07-26:
What os?

ABRT29, 2020-07-26:
Android devices mustly

mattkhaw, 2020-07-30:
To answer your question, in short, you can't achieve what you want to achieve if you are using this plugin only. It must be complemented with another plugin. Try looking up cordova-plugin-beacon (I think that's the name). You need to combine this plugin with that in order to work in background mode.

ABRT29, 2020-07-30:
hello, thanks for your help. 
Indeed with the beacon plugin it works perfectly but in the background the application gets kicked out after 2 hours

mattkhaw, 2020-07-30:
That's a problem indeed. Not sure mine will have this kind of issue or not. Why not replacing the interval to beacon based? I think this should not pose a problem since it will only do whatever you want to do when your devices come into contact with one another. I'm using this method in one of my apps. I haven't tried to run it for 2 hours. Maybe you can try that and let us know about it.

ABRT29, 2020-07-30:
sorry i forgot to specify that i no longer use interval but i am using beaconregion and startMonitoringForRegion and this is how my app gets killed after two hours. it's much better than the 5 minutes of my previous tests but not enough for the use I want to make of it

randdusing, 2020-07-31:
This plugin definitely supports running in the background. The app will "restore" on certain events like subscribe notifications.

mattkhaw, 2020-08-03:
@randdusing yes, the current plugin works in background and the callbacks are only called when the app is resumed. Our primary logic is on Javascript code and our code doesn't run until the app is resumed. This pose a problem for some apps which require constant monitoring in background. I do think that there's a way to achieve this. I've opened an issue for this as well.